vlachoudis/bCNC #510: Implementation of new GRBL Jogging Feature ?

tmie, 2017-02-01:
Any plans to implement the Jogging with GRBL 1.1 '$J=' command?
Seems interesting:
[Grbl v1.1 Jogging](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging)

Effer, 2017-02-01:
I don't remember where but @vlachoudis said is going to implement it soon.
Briefly reading the doc it seem not so straightforward. Lot of test and debugging ahead I suppose.

tmie, 2017-02-01:
Ok, I will be patient.
Thanks for the answer :-)

closed 2017-02-01 by tmie

hdo, 2018-09-26:
Any updates on this yet? Up to now the new jogging feature has not been implemented yet :-(

PCSDIAS, 2018-10-15:
With the patch shown on August 8 I have got this functional cellphone bCNC GUI.
[image]

PCSDIAS, 2018-10-15:
Hi all.
I'm sorry to tell that I misplaced the last comment.

The correct should be on "Jogging bCNC with a cell phone #930"