vlachoudis/bCNC #511: Change colors with global settings

DeerJohn, 2017-02-03:
This is not about canvas color! This is about colors of interface elements.

I have something wrong on my system (with my favorite color scheme): white text on white background. I try to change it, but background color set explicitly in many places in many files (*.py). I do this work as I can (see patch) and add new setting for [Color] section: global.background. Yet can i change color by editing only one place. I think, that other explicit color declarations (foreground="DarkBlue", etc.) must be changed to global settings.

[bCNC-0.9.7-add-global-background-variable.patch.txt](https://github.com/vlachoudis/bCNC/files/750208/bCNC-0.9.7-add-global-background-variable.patch.txt)

PS: I don't known Python/Tk enough, so my patch not optimal, but it work.

vlachoudis, 2017-02-05:
Good suggestion. I will try to integrate it

funklord, 2018-02-25:
I have the same problem and the patch is now broken.

ptvoinfo, 2019-09-23:
I have the same problem on Ubuntu 18.04 and default 'dark mate' theme. All controls have white text on the white background.

heschlie, 2020-06-01:
Any updates on this, love bCNC but not a lot of fun to need to change my theme to use it :cry:

boris-so, 2020-11-21:
> Any updates on this, love bCNC but not a lot of fun to need to change my theme to use it

It seems the patch has been applied. Navigate to file `/home/user/.local/lib/python3.7/site-packages/bCNC/bCNC.ini` line 223 and u should see a set of empty options:
```
global.background =
global.foreground =
global.activebackground =
global.activeforeground =
global.disabledForeground =
global.highlightBackground =
global.highlightColor =
global.selectBackground =
global.selectForeground =
```
Try to change them and check.
Worked for me on kde 5.14.5 and bCNC 0.9.14.308