vlachoudis/bCNC #513: Error 33, GRBL 1.1

Guenni75, 2017-02-06:
I use the latest bCNC Master and some times i get an Error 33.

Terminal:
G1X214.0735Y15.5362Z-4.8359
ok
G1X214.1053Y15.4479Z-4.845
ok
G1X214.1099Y15.4374Z-4.8462
ok
G1X214.1053Y15.4479
ok
G3X215.5Y14.5I1.3901J0.5626F300
error:33
Run ended
2017-02-06 16:02:20.876000
Current: 1129 [1322]  Completed: 85% [16m57s Tot: 19m51s Rem: 2m54s]

I think it is the J in the G3 Code.

I dont know, if it is a problem with GRBL 1.1 or with bCNC.

[image]

[Kontur 3mm.zip](https://github.com/vlachoudis/bCNC/files/754996/Kontur.3mm.zip)

I never had something like that befor.

vlachoudis, 2017-02-06:
It might be due to a patch I reverted yesterday. I received it sometime ago to ignore errors when not running. Unfortunately the condition was not properly checked and resulted in ignoring the errors of grbl instead of stopping. When you have the error click on the red error label to get information on the error as well the error line.

Guenni75, 2017-02-07:
I searched the web and it is a problem with GRBL.
This is not a new Problem. I found posts 2 years old, describing this error.
GRBL has sometimes Problems with arcs (G2 and G3).

Look at this: [https://github.com/grbl/grbl/issues/619](https://github.com/grbl/grbl/issues/619)

closed 2017-02-07 by Guenni75

Giochia, 2020-05-11:
Hello, I get "Error 33" in a file generated by bCNC itself!
It is a file generated by the Text generator in the Tool menu. I cannot remember the name of the font file, but it worked fine a couple of versions ago. Now, by using the same file I get an error and I cannot finish the job.

These are the last lines that appear on the "Terminal" view: 

"G2X108.197Y0.025I-0.888J1.468
ok
G2X107.913Y0.009I-0.402J4.629
ok
G2X107.221Y0I-0.692J-0.009
error:33"

Please can anyone help?

[image]

MARIOBASZ, 2020-05-11:
Maybe that and other plugins should be fixed.
If the rounding of i j k is not performed upward, error 33 may occur (that arc cannot reach the target point.