vlachoudis/bCNC #515: Camera selection ?

tmie, 2017-02-08:
Hello,
I use bCNC on my Laptop and want to use a USB camera to find the zero point of my workpiece.
Unfortunately I see the picture of the camera built in the laptop instead what the mill camera sees.
How can I switch to the external USB Webcam?
Thanks in advance
Tom

Effer, 2017-02-10:
Hi @tmie,
I think you should modify the **.bcnc** file manually. The file is located in your **user** folder an **not** in the bCNC folder. (sorry for the bold but often users take the wrong file)
Then you should add:  
webcam = 1

under the [Camera] section. Something like this.
``` ini
[Camera]
aligncam_d = 0.0
aligncam_dx = 0.0
aligncam_dy = 0.0
aligncam_z = 0.0
webcam = 1
```

I didn't test it..

tmie, 2017-02-11:
Hello,
thanks for the suggestion. 
Unfortunately it did not work, I still get the picture of the built in camera with "webcam = 1".
I forgot to tell some details:
Xubuntu 14.04 LTS (at this moment updating to 16.04LTS) , 64bit
Function of the external cam tested with VLC
Internal camera is /dev/video0
External camera is /dev/video1
Any idea what i did wrong?
By the way: is there any documentation where I would find hints like yours what I can do in which file? Sorry, I can't find the needed informations on my own :-(
Greetings from Germany
Tom

vlachoudis, 2017-02-12:
webcam = 1   (for the web camera)
aligncam = 1 (for the camera used for alignment)

tmie, 2017-02-13:
Sorry, but this did also not work, same result: the image of the internal webcam :-(

vlachoudis, 2017-02-14:
It could be that your external camera is not recognized by opencv

lalo-uy, 2017-02-14:
Are you sure that the camera is not taken by other program?

lalo-uy, 2017-02-14:
Here:
http://stackoverflow.com/questions/7322939/how-to-count-cameras-in-opencv-2-3

are some ways to check the number of cameras that OpenCV detect.

Hope is usefull to you.

2017-02-14 12:12 GMT-03:00 Eduardo Spremolla <e.spremolla@gmail.com>:

tmie, 2017-02-14:
Thanks for your answers, today I tried it again (with the third different camera) and it worked!
After this I plugged it out and connected the camera I want to work with and it still is ok.
I have no idea what is different today (already shut down and restarted the laptop several times the last days).
Again, many thanks for the answers and for this great project !
Tom

closed 2017-02-14 by tmie