vlachoudis/bCNC #517: infinite loop

greyltc, 2017-02-09:
Thanks for the really cool tool!

I was working on generating a toolpath today and I've come across a set of operations that results in the program entering an infinite loop:

1. [Import the attached, very simple DXF](https://github.com/vlachoudis/bCNC/files/764096/bugReport.dxf.txt) (can only attach .txt, rename to .dxf) 
[image]
1. Set your endmill to have a 3.175mm radius
1. Perform a "Profile" operation on the loop
  -Direction: Inside
  -Offset: 0.5
  -Overcut: True
1. Perform a "Pocket" operation on the profile from the previous step
1. Watch your cursor spin forever

Just now, I used the built in error reporting function to submit an error report about this.

vlachoudis, 2017-02-11:
Thank I will try to debug what is wrong

vlachoudis, 2017-02-11:
I've tried your dxf and your instruction and I cannot reproduce it. However if you try consecutive profile operations after a certain point it loops since it confuses what is in and what is out due to the overcuts. 
Most probably your infinite loop comes from there. I will try to patch it.

Harvie, 2019-03-04:
Do you still have this problem in latest git master? Otherwise i will close the issue.