vlachoudis/bCNC #518: Change the .bat

Majestad278, 2017-02-10:
Sorry for this, but the .bat is something anti-aesthetic for me, so I created an .exe without the windows consoles (CMD) or the Python console frame, because I do not like having open things that I do not use and annoy the view , And I have the impresion, since I use the .exe the bCNC opens faster.

P.D: I do not know if this goes in issues or in pull request :)
P.D.2: I'm sorry for my English but I'm Spanish and I do not speak it well

Thanks for your time, regards Majestad278

vlachoudis, 2017-02-11:
I am not a windows user, so I don't know how to do this. What are the limitations of the exe vs the pure python code?

vlachoudis, 2017-02-11:
The .bat is not really needed. You can start directly by clicking on the bCNC.py
Also if you replace the association of .py with pythonw instead of python it will not open any console.
However the console is useful at least for debugging :)

Majestad278, 2017-02-11:
Yea thats true, I don't want to criticize your work, It's a fantastic program, but I only mean the .exe seems more professional to the end user.

Well is my opinion, I apreciate your repply, If you change your mind tell my and I send de .exe to attached at the code.

Thanks for all, best wishes and luck

Effer, 2017-02-11:
Time ago I've ported bCNC to be a portable application for windows. The advantage there is that python is not required any more. Maybe we could add that to the repository. It's still here #437 if you want to have a look.

Majestad278, 2017-02-12:
Thanks for the info, I gona see the threat when I has some free time :D

vlachoudis, 2017-02-12:
Why not @Effer to offer it as a branch

Effer, 2017-02-13:
@vlachoudis that sound good.

closed 2017-02-14 by Majestad278