vlachoudis/bCNC #523: drill dwell is misplaced

greyltc, 2017-02-15:
I've made some drill operations with:
Target Depth = -2.7
Peck depth = 1.0
Dwell (s) = 3.0
Distance = 
Number = 1

The gcode generated here contains:
```
g1 z-1 f50
g0 z3
g4 p3
g1 z-2 f50
g0 z3
g4 p3
g1 z-2.7 f50
g0 z3
g4 p3
```
It looks to me like the pause is in the wrong place (should be after the plunge, not after the retraction). Does anyone want the peck operation to dwell after the drill is retracted rather than at the bottom of the peck?

onekk, 2017-02-15:
One "school" says that the scope of the dwell is to clear the drill flutes from the cuttiing debris, and if it is done with the drill in the hole (prior the retraction) it will not clear the drills flutes.

But there is another "school" that say that a small dwell is necessary at the bottom to clear the hole prior to peck up to facilitate the clearing of the flutes.

Both "schools" says that the retraction height must not leave the hole as in this case the debris could re-enter the hole causing a premature wearing of the drill bit.

greyltc, 2017-02-15:
@onekk, that's a really great point about the retraction height. Here it looks like the peck cycle always returns to my "safe" height (z=3).

I suppose it would be best if we could somehow configure everything; bottom dwell time, top dwell time and retraction height.

onekk, 2017-02-16:
As the "canned cycle" is expanded in bCNC,  as GRBL dind't support them, I suppose it can be personalized in some way. 

But how do you produce the GCode, maybe the CAM program could expand the cycles itself and personalize all their aspect.
In a CAM program if you set a "drilling operation" maybe there is some options to set some preferences and then the CAM itself will produce the basic GCODE for the canned cycle without emitting the G8x code and only G01 and G0 and G4 in the appropriate places.

I use CamBam and there is some place to personalize the drilling operation, as I didn't have a professional CNC to make the holes I use the "spiral drilling" method if the hole diameter is lower than 2 times the bit diameter, and set a pocket operation if the hole diameter is greater than 2 times the bit diameter (to avoid the leftover pin to destroy the hole and maybe the bit).

Regards