vlachoudis/bCNC #525: Camera doesnt works on Windows

Fosforitofercho, 2017-02-20:
I have a camera and want test the proove , but it doesnt works , i push the button of camera settings and nothing happen ,only active the CNC settings ,rare . as shown on image 

[image]

After y run the code tah you suggeted in other issue 

import cv2
camera = cv2.VideoCapture(0)
camera.read()

and the camera LED indicator turns on 

Now i tried to fix the problem , inspect the code in file  "Camera.py" , but it has been imposible follow the problem, because i dont know how debug the problem or some type  o log on the program .

Two Things , one  please help me fix the problem , second  a way to make debug , prints on console ,log file  , even popup windows 

I use WIndows 7 x64 , phyton and all packages 32 bit  and bCNC 0.9.7

Thank you so much

poesel, 2017-02-23:
Have you installed OpenCV, PIL and imageTk for python?
Is the camera working in other applications?
To switch on the camera in bCNC you need to click the small camera icon in the small menu bar just to the right of the blue CNC box in your screenshot (maximise window to see it).
The camera configuration button does nothing for me either but it works anyway (Linux).

Fosforitofercho, 2017-02-25:
[image]
 Yes, I have installled the modules, but it doesnt works  :S , the camera button is that you see in the red box . ¿Can  you send me a video setting this perfect, i cant find some on youtube?

poesel, 2017-02-25:
No, the camera button is not in the red box. That is camera configuration.
Maximise bCNC and go straight(!) from the blue box to the right until you find a small camera symbol.
That is the camera button.

Fosforitofercho, 2017-03-01:
juputa , , now it works, OMG , it is not clear on the tutorial, incredible !!!

closed 2017-05-11 by vlachoudis