vlachoudis/bCNC #526: Editor Gcode parsing breaks editor

amateurengineer, 2017-02-22:
Hello,

when I import a gcode made from a pcb design made in KiCad, then translated from gerber to gcode with pcb2gcode the Editor falls into weird behavior:

- The scaling of the object is correct
- If I move an object 10mm by mouse it will be moved by 2,54 * 100 in that direction.
- The Predefined move tools also fail in a similar manner (the to move direction overshoots by that value).
- Objects like bowl are too big by 2,54 * 100

In the gcode file header is the G20 statement

```
G94     ( Inches per minute feed rate. )
G20     ( Units == INCHES.             )
G90     ( Absolute coordinates.        )
S30000  ( RPM spindle speed.           )
M3      ( Spindle on clockwise.        )
```

when it is removed like

```
G94     ( Inches per minute feed rate. )
G90     ( Absolute coordinates.        )
S30000  ( RPM spindle speed.           )
M3      ( Spindle on clockwise.        )
```

- the editor is just fine but the objects scale is off by 100

I hope this wasn't reported already, thank you for your work.

vlachoudis, 2017-02-24:
Have you configured bcnc to inches in the config?

amateurengineer, 2017-02-24:
Hello,
Units (inches) is unticked.

I have compiled pcb2gcode v1.2.3 from source to make it accept metric-output option which works around the problem.