vlachoudis/bCNC #527: manual step distance everywhere

poesel, 2017-02-23:
I'm starting to use your excellent camera probing feature. There you have to move the gantry around a bit. Sometimes fast, sometimes slow. I'm using an external keyboard for this.
Problem is that you can't see the current step distance when you are in camera mode. So you might get some unwanted movement if you are not careful with changing the steps.

If the current step distance would be shown somewhere all the time that would be nice. Thank you.

Effer, 2017-02-25:
Hi @poesel ,
even if not so user frendly, bCNC interface can be custimized to your needs.
First give a look at the [wiki](https://github.com/vlachoudis/bCNC/wiki/Customize%20Menu), there you'll find how to customize the ribbon menu (top menu). The same can be applied to the "page menu" left side part.

In short, add this line to you **.bCNC** user file in the [bCNC] section:

**probe.page = DRO ProbeCommon Control**

to have the control in the [Probe] section too:

[image]

Hope this help

poesel, 2017-02-25:
Cool - that is very useful. I've also added the run command in the probe ribbon as per wiki.
IMHO these option should be the standard.

Thanks!

closed 2017-02-25 by poesel