vlachoudis/bCNC #529: Somthing wrong when i load a gcode file.

asm7100, 2017-02-24:
When i load a file, its looks like it outside the machine boundaries.
My CNC table has x0 y0 at the lover left corner but when i load something its place the drawring in the top right corner. How to fix this?

Effer, 2017-02-25:
Hi @asm7100 
probably you need to set maximum travel distance in your config ( from Tools Tab in bCNC software).
See the [wiki](https://github.com/vlachoudis/bCNC/wiki/Configuration).  Travel x/y/z section.
Be aware: for historical reasons values are always referenced **from top/right corner** even if your home switch position is in the bottom/left corner.

asm7100, 2017-02-25:
Its all set :-)
Is the a way to tell it that home is Bottom/left (or disable the "layout" of the boundaries?)

Effer, 2017-02-25:
Using negative travel distance.

Effer, 2017-02-25:
You can also disable the workspace draw with the button on the ribbon over the canvas. Hover with the mouse to see tooltips.

closed 2017-03-06 by vlachoudis

asm7100, 2017-03-13:
Now the Y axel is ok, but not my X axel, it dosent matter if i put a minus in front of it, its always at the right of my drawing..
And X is - on the right site of my XY zero point.