vlachoudis/bCNC #530: Can't unlock controller after hitting limitswitch. It occurs on a simple gcodes.

shateks, 2017-02-27:
When bcnc is sending simple gcode, and grbl is in half way to finish. I'm hitting limit switch. Before hitting limitswitch i see that all gcodes ware send, becouse all lines in viewer are colored green. Then bcnc raports alarm 1, I can send software reset, but I can't unlock device. Only solution is to perform hardware reset.
I have checked in usb/serial sniffer, bcnc in fact is sending 0x18 (softreset), but not sending $X (unlock).

This issue happens only on very simple gcodes, for instance:
M3 S12000
G4 P3
g0 z3
g0 x0 y0
g1 z0 f200
f380
g1 x0 y10
g1 x10 y10
g1 x10 y0
g1 x0 y0
g1 x0 y0
g0 z3
M5
G0 Z20

The issue is not happening when gcode i bit more complicated, and before hitting limit switch not all gcode lines in viewer are green colored.

bcnc version 0.9.8
grbl version 1.1f

vlachoudis, 2017-03-06:
I don't understand the problem? If you hit the limit switch the controller remains locked and you need
to send soft-reset and then unlock. bCNC is not sending the unlock by it self.
Also note that grbl in the past when a limit switch was active you could still send a move command after a reset, not any more in v1.x

shateks, 2017-03-07:
Perhaps it's only happening on my hardware/software. But I encourage bCNC users to confirm or reject this issue. Of course in specific conditions described above.

> bCNC is not sending the unlock by it self

Yes I know that bCNC is not sending unlock by itself, a was clicking reset then unlock. And as I said, 0x18 was send, $X was not. Observed in Wireshark on USB.

closed 2017-04-22 by shateks