vlachoudis/bCNC #533: bCNC can add a simple mode of operation

viewsat, 2017-03-06:
Recommended bCNC can add a simple mode of operation, on the right only show send G-code, like the following picture.
[image]

vlachoudis, 2017-03-06:
It could be a possible enhancement. You have anyhow this information in the "Terminal' which is synchronized with the replies of grbl. Not in a table format.

kworsfold, 2017-10-29:
Agreed.... This would help in debugging as it would show the current line when it errors

vlachoudis, 2017-10-30:
Actually in my TODO list was to make a step by step debugger. That you go in the editor and sending individual commands or blocks to grbl, and inspecting the state.... I need to find some time to do it.

kworsfold, 2017-10-30:
That would be cool.... very useful making problems easier to resolve