vlachoudis/bCNC #535: Load gcode through web pendant

moromusa, 2017-03-09:
Hi everbody,
I am using bCNC with RPi and a touchscreen. My intention is create a small program that to control my CNC through bCNC's pendant function. 

The idea is execute bCNC and a web browser in background, and the program as a simple interface for the touchscreen. I write it in the url bar from the browser, using it as the command line via the [/send?cmd=] function. It works well for sending simple gcode, movements... but fails when loading the gcode file. 

I write this in the url bar: localhost:8080/send?cmd=load (file directory). Then the file is loading endless (loading... in the left below corner of the program). If I try to load the file with the upload button, the file loads, but loading process don't end too (as in a loop). However, if I try to write "load (file directory)" directly in the program CommandLine, works perfectly, loaded file message appears.

Any ideas? Or maybe there are an easiest way to do that. 
Thanks a lot

Effer, 2017-03-09:
Hi @moromusa ,
I've just try to load a gcode file from the web pendant.
Try to use a double slash for the path `http://localhost:8080/send?cmd=load d:\\test.ngc`
It works for me on Win10.

Sorry but I'm curious to know why you're using the browser url bar to send command.  I believe any language out there can send some simple web command from code.
That said there's also "select file to upload" way. That webPendant load function works using a POST  with url  /upload . It acts a little different then the "load" command from CommandLine as it's used to upload file from another computer to bCNC (in case you need it) and not only loading file already present in bCNC.

moromusa, 2017-03-09:
Hi @Effer, thank you for your answer.
I solved this issue. The problem was the name of the folder contains a special character [à]. Change the gcode file to other folder and it works, with single and double slash.

Sorry for that, I'm newbie in programming, and I find this way to comunicate a secondary program with bCNC, as mentioned on issue #155 . I am using the url bar just for testing on Win10 and Firefox. The idea is create some simple program to send this comands from code (as a web adress?). Just as you said.

Effer, 2017-03-10:
There was some discussion on how integrate a channel to speak with bCNC but not much more. The web interface command is something raw but that works.

closed 2017-03-18 by vlachoudis