vlachoudis/bCNC #536: Z position at amera alignment

Rudi84, 2017-03-09:
Hi,

I have a small problem with the camera alignment. When I switch between drill and camera, x and y is moved correctly. But z is always driven zero heighth. This costed me already a lot of bits, because sometimes z zero is still in the material. Is there an option, where Z staies at the heighth where is it, for camera switching?

vlachoudis, 2017-03-09:
Z is taken as the location of the spindle when you registered the camera, and it is written under in the Camera frame "Offset:" the 3rd parameter. The first two are the relative DX, DY spindle-vs-camera and the third on is the absolute of the camera.
The problem with the camera alignment is that if the camera is not perfectly aligned with the Z axis any mismatch in the z-height will create an artificial offset in x,y. This is the reason I am moving it a the camera height when it was registered.

Rudi84, 2017-03-09:
Thanks for the fast response. I understand your intention. But I did this alignment for my cam, because depending on the length of my bit the distance of cam to worksheet is more or less. Is there any line in a file, where I can modify this for me?

vlachoudis, 2017-03-09:
You can alter the "Offset:" third parameter. I could potentially put a check that if the
field is empty to ignore and not do any z-movement.

Rudi84, 2017-03-09:
Yes, alter the offset will work, but I am sure I will forget in a while, break the bit again. So that check would be perfect.

vlachoudis, 2017-03-10:
I've added a protection that if the camera Z is empty to move only on X&Y. Could you test it

vlachoudis, 2017-03-10:
Just remember that when you enter a value by hand you need to press "Enter" to update it

Rudi84, 2017-03-10:
Thanks a lot, I will report next week. Is there away to support you with a contribution via paypal or something like this?

vlachoudis, 2017-03-10:
Thank you @Rudi84 I don't have anything for contributions. Now that you mention, I would strongly prefer any contribution to be send for charity purposes. I am supporter of the organization (Smile of Childs) in Greece https://www.hamogelo.gr/gr/en/stirikste-mas/  a non-government organization to help kids. Maybe I should add it on the wiki.

Rudi84, 2017-03-13:
I tested, and it is working perfect. I will follow your recommendation. Thanks

closed 2017-03-13 by Rudi84