vlachoudis/bCNC #538: Moving the viewable area in the canvas with 3rd MB sometimes moves the carriage

Masc99, 2017-03-14:
Hi,

I'm using BCNC on a Raspberry Pi 3 with Raspbian.  Sometimes when I move the viewable area in the canvas with the third mouse button, the motors start and move the carriage. Unfortunately this in not really reproduceable and happens only rarely.
I'll try to gather some more information when it occurs again.

thx

masc

vlachoudis, 2017-03-15:
Strange. Just for information that if you press the "G" button before clicking the mouse it has as an intention to move the gantry. Also, please look at the terminal to see what command was send to grbl.

Masc99, 2017-03-17:
Yeah, I know that "G"-feature, but it looks and sounds more like if the carriage would follow the mouse-drag movement. I will investigate this further and try to reproduce it somehow or at least find some clues.

Masc99, 2017-03-19:
Yesterday I had the issue again. I just dragged the viewable area over and over again and all of a sudden the carriage moved. A look into the terminal showed, that a jog-command (moveXdown) was performed. Luckily the stepwidth was set to 1mm so nothing crashed into the limit-switches. ;)
When I had it like this, I didn't need to move the mouse very much. While the 3rd mb was pressed, every little mouse movement resulted in

G91 X-1
ok
G90
ok

output in the terminal. So obviously the ControlPage.moveXdown-function is called. It would be nice to see from where it is called. Is it a GUI-event or a keaboard-event? I think both trigger the same function, right? (And no, I'm not clicking buttons or hitting keys on the keyboard...)
I will add a "traceback.print_stack()" to the function, try to reproduce it again, and see it outputs something to work with.

vlachoudis, 2017-03-20:
This is exactly what I wanted to propose. Also put a print event, since the keyboard week have the event filled but not the button press.

Harvie, 2018-10-29:
does this still happen with latest master?