vlachoudis/bCNC #541: Scan-function doesn't work correctly when "incremental" is set

Masc99, 2017-03-17:
Hi, yesterday I tried to use the scan function the first time and it moved completely wrong. Checking the sent GCode in the Terminal, I realized that the X/Y-coords where absolute coords (as in the loaded-file) but the machine was set to incremental-mode (maybe from a previous jog-command). 
After settings the machine to abs-mode, scanning worked fine.
So I think the easiest solution would be to send G90 and update the GUI before sending the scan-coords.

thx 

masc

Masc99, 2017-03-17:
I just found the [Buttons]-section in bCNC.ini
I will add the G90 there and test it tomorrow. Seems as it should do the trick.

vlachoudis, 2017-03-18:
Indeed. Thanks for spoting it.

vlachoudis, 2017-03-18:
done

closed 2017-03-18 by vlachoudis