vlachoudis/bCNC #542: [Request] Add Option to apply auto-level information to loaded code

Masc99, 2017-03-17:
Hi,

it would be great if one could apply the generated auto-level-information to the loaded gcode. This would serve several porpuses:
1. Chicken mode: You can check the "new" gcode before sending and see if everything looks reasonable (Useful when you get confused with TLO... ;) )

2. Production mode: This comes in handy when you want to reproduce cuts on uneven surfaces:
- create the design
- create the height-map
- apply to the gcode
- save it

I think this wouldn't even be a lot to do as most of it should be the movement of code and adding an apply-button.

vlachoudis, 2017-03-18:
@Masc99 this is relatively "easy" to do, and it could help as well debugging in case of problems.

vlachoudis, 2017-03-22:
Done on the latest master. You can apply the autolevel information in any selected block of the code, from the "Editor -> Edit -> Autolevel". Then you can save it normally as gcode in any file.
Warning if you apply the autolevel information you have to clean the autolevel from the "Probe" otherwise if you run it, it will be applied twice. I was somehow tempted after applying it to delete the autolevel information, just to avoid this problem, but I am not sure what can be the best option.