vlachoudis/bCNC #544: DRO (both machine and work position) and status aren't updating

FishyD, 2017-03-19:
Windows 10, X-carve w/ X-Controller, GRBL 1.03C (X-Carve fork). 

bCNC connects the machine (COM3) but the status and both coordinates aren't correctly updating. I can unlock the machine (even though the lock icon always shows unlocked) with $X or home but as I jog the machine the coordinates aren't moving, they always read 0,0,0 or sometimes all 6 are blank. I have attached a screen shot, as you can I have a connection but the status says "not connected" and there are not machine or work coordinates.  Sometimes new errors will populate; I forced a homing fail, but still no correct DRO readout. 

[image]
[image]

digicnc, 2017-03-21:
I have the same problem, my GRBL Version 1.1e, with the terminal I can check the connection, but the DRO Section showing "Not connected"

FishyD, 2017-03-21:
Exactly! I can check the connection, pull the $$ and jog but the DRO shows no connection and doesn't update the coordinates.

vlachoudis, 2017-03-21:
Can you send a copy of the terminal

digicnc, 2017-03-22:
Here a copy of the terminal:

[image]

If you want that I run a specific command, just let me know. 

Thanks

vlachoudis, 2017-03-22:
Can you set $10=1

digicnc, 2017-03-23:
Ok, I tried, this, now it is working alright ....
Thanks a lot

vlachoudis, 2017-03-23:
You are welcome

closed 2017-03-23 by vlachoudis

FishyD, 2017-03-24:
Unfortunately even with $10=1 (it was at $10=2) I am still not getting position reporting on the DRO. See my terminal below. 
[image]
[image]