vlachoudis/bCNC #545: bCNC file size restriction on RPi

Lylesgold, 2017-03-22:
bCNC works very well for me on an RPi3 for small files but for a larger gcode file (807Kb which I'm told is not particularly big) seems to be capacity limited. It accepts the file completely (all present in bCNC editor) but the graphic simulation is incomplete, showing only a little over half of the expected paths. (the same file works fine on bCNC under Windows10). As this looks like a memory allocation issue I increased the RPi GPU allocation up to 832Mb without improvement. Sonny Jeon (Openbuilds forum) was kind enough to look at this and thought it may be a matter of bCNC memory allocation with Python. Has anyone else experience of this? Is there a solution, is it in bCNC or on the RPi side? 

I'm handy, but don't have the knowledge to diagnose/fix this. Love using bCNC, especially over VCN and looking forward to moving past trinkets/test pieces onto real projects. Thanks.

lalo-uy, 2017-03-22:
Ther is a time limit while rendering the image. try to rise it.

Lylesgold, 2017-03-23:
Makes sense. Different processor speed may then explain why the Windows machine drew all paths but not the RPi. Finding and altering the time limit is beyond my ability. Won't this require a new bCNC build?

electrokean, 2017-03-23:
@Lylesgold there is a dropdown at top right to select the drawing time limit - see https://github.com/vlachoudis/bCNC/issues/485

Lylesgold, 2017-03-23:
That's it! Thanks, making this parametric saved the day. I had to extend the time limit to 20 seconds on the RPi, but the path is now fully rendered. As a new user I really appreciate the help you are all providing and although I don't know how to access the status line Vlachoudis refers to in #485, I know what to do in future.

closed 2017-03-23 by Lylesgold