vlachoudis/bCNC #546: If I repeatedly press PAUSE bCNC  generate an error and stop working!

viewsat, 2017-03-23:
When I was sending gcode, if I repeatedly press PAUSE, then bCNC will generate an error and stop working
!

[image]

vlachoudis, 2017-03-23:
Here it seems, that the serial is failing. Normally the "!" should be consumed immediately by grbl, so it should not matter how many you push on the grbl buffer. So either is a hardware error (noise on the line), or for some reason grbl is not removing the "!" from the buffer when in Hold state. @chamit any other possible explanation?

chamnit, 2017-03-23:
@vlachoudis : The '!' character should always be intercepted when received by Grbl. I would need to know more about what the failure is to determine if Grbl is the culprit.

vlachoudis, 2017-03-23:
@chamnit python is reporting timeout on serial write "!". The only time I saw this exception on hardware errors.

chamnit, 2017-03-23:
@vlachoudis : Ok. I don't see how Grbl could create a write error on the host side since it just receives the unsynchronized bi-directional data. I would agree that it's likely hardware related or electrical in nature. Also make sure you are not using an Arduino with a CH380 USB-serial chip. These are known to have communication issues.

I just read a thread this morning about a very similar identical issue. It was due to a noisy electrical environment with a spindle under load on the same circuit as the controller and host computer. It was solved by placing the Grbl and the host computer behind a battery backup unit (UPS), while the spindle was connected to the wall receptacle. UPS's have a lot of buffering electronics in them that seemed to help prevent voltage spikes from interfering with communication.

viewsat, 2017-03-24:
I tried orange pe PC + Armbian last night to perform bCNC, and there seemed to be no mistake.
Arduino with a CH$340G USB-serial chip.

[image]

[image]

viewsat, 2017-03-24:
But when sending gcode, such as changing the spindle% will have an incorrect action.

viewsat, 2017-03-26:
I send about 1 million lines of gcode, no problem, except spindle override has occurred wrong action.
use orange pi PC & Armbian .

[image]

closed 2018-10-29 by Harvie