vlachoudis/bCNC #547: .tap file not milling

danielpcb3d, 2017-03-23:
Dear all, i'm a newbie in bcnc. Some weeks ago I've started building a pcb milling maschine. I use the following components/projects:

- PCB cyclone (https://github.com/CarlosGS/Cyclone-PCB-Factory)
- Protoneer RPi CNC Hat (http://wiki.protoneer.co.nz/Raspberry_Pi_CNC)
- DRV8825 (1/32 Microsteps)
- 1,8° Steppermotors

The manual control of every axis works fine only the .tap files (eagle) didn't work. What i'm doing before starting bcnc: I load the .tap file into the workspace, press x,y,z = 0 and after that i press start, but the motors only making some strange noise but they didn't rotate. 

Does someone of you had the same problem and could tell me what i've to do to solve it?

electrokean, 2017-03-23:
Sounds like the motors are stalling.  Try reducing your max step rate and acceleration settings.

danielpcb3d, 2017-03-24:
Thanks a lot for your reply. Could you please tell me where i can reduce it?

electrokean, 2017-03-24:
In your grbl configuration of the Protoneer board - see https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration for details of the settings.  You want to reduce $110,$111,$112 and $120,$121,$122 - especially the latter ones which are acceleration.  Keep halving them all until the motors move, then tweak them up/down to get the best performance while avoiding stalls.
From bCNC I think you can do access the grbl settings via the terminal window - I don't have a copy running here right now to check.

danielpcb3d, 2017-03-24:
Thanks a lot for your help. I will test it on this weekend.

vlachoudis, 2017-03-24:
I am closing the issue since it is not related with bCNC

closed 2017-03-24 by vlachoudis