vlachoudis/bCNC #548: Fusion 360 generated code stops just after start, but runs ok in universal gcode sender

stasjminister, 2017-03-24:
As mentioned I am having issues with 2 files generated in fusion360. But both run ok when sending with universal gcode sender 0.9. 

Right after pressing start the machine moves a bit, then suddely halts. With no error codes that I can see..

Any suggestions?

I am running bcnc 1.0.9.

GRBL code in txt-file. (had to change extension from.nc
[dobbel sirkel4mm flat.txt](https://github.com/vlachoudis/bCNC/files/869336/dobbel.sirkel4mm.flat.txt)
)

atrueresistance, 2017-03-24:
Generate the code with no coolant and it will run fine

stasjminister, 2017-03-24:
Tanks! Will try asap. 3.14 thumbs up for extremely quick reply!

stasjminister, 2017-03-25:
I also had to change the post processor in fusion360 to the carbide3d one. 
With the default encoder I got lots of error 33 and error 36 as well when I finally got it run, after 5-10 presses on start. So I guess the default encoder left a whole lot of gravel for bcnc to chew trough.

Thanks for your help. Much appreciated. If your ever in Oslo, Norway, I owe you a beer :)

closed 2017-03-25 by stasjminister