vlachoudis/bCNC #549: Terminating app due to uncaught exception 'CALayerInvalidGeometry'

ndevenish, 2017-03-25:
I seem to get (and have done with master, v0.99 and older versions), reasonably constantly, the following error:

```
2017-03-25 16:16:43.076 Python[75083:2275168] *** Terminating app due to uncaught exception 'CALayerInvalidGeometry', reason: 'CALayer position contains NaN: [0 nan]'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff8ff606fb __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fffa4767a2a objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff8ffdd9a5 +[NSException raise:format:] + 197
	3   QuartzCore                          0x00007fff95b28520 _ZN2CA5Layer12set_positionERKNS_4Vec2IdEEb + 152
	4   QuartzCore                          0x00007fff95b28695 -[CALayer setPosition:] + 44
	5   QuartzCore                          0x00007fff95b28ceb -[CALayer setFrame:] + 644
	6   CoreUI                              0x00007fff9b8fd4ce _ZN20CUICoreThemeRenderer26MakeOrUpdateScrollBarLayerEPK13CUIDescriptoraPP7CALayer + 1284
	7   CoreUI                              0x00007fff9b8f94c3 _ZN20CUICoreThemeRenderer19CreateOrUpdateLayerEPK13CUIDescriptorPP7CALayer + 1595
	8   CoreUI                              0x00007fff9b87a58d _ZN11CUIRenderer19CreateOrUpdateLayerEPK14__CFDictionaryPP7CALayer + 175
	9   CoreUI                              0x00007fff9b87d241 CUICreateOrUpdateLayer + 221
	10  AppKit                              0x00007fff8e58c21f -[NSCompositeAppearance _callCoreUIWithBlock:options:] + 226
	11  AppKit                              0x00007fff8dc320b3 -[NSAppearance _createOrUpdateLayer:options:] + 76
	12  AppKit                              0x00007fff8deaa71f -[NSScrollerImp _animateToRolloverState] + 274
	13  AppKit                              0x00007fff8de6a0f9 __49-[NSScrollerImp _installDelayedRolloverAnimation]_block_invoke + 673
	14  AppKit                              0x00007fff8dd302c5 -[NSScrollerImp _doWork:] + 15
	15  Foundation                          0x00007fff9195ec88 __NSFireDelayedPerform + 417
	16  CoreFoundation                      0x00007fff8fee0e84 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
	17  CoreFoundation                      0x00007fff8fee0b13 __CFRunLoopDoTimer + 1075
	18  CoreFoundation                      0x00007fff8fee066a __CFRunLoopDoTimers + 298
	19  CoreFoundation                      0x00007fff8fed8231 __CFRunLoopRun + 2081
	20  CoreFoundation                      0x00007fff8fed77b4 CFRunLoopRunSpecific + 420
	21  Tcl                                 0x000000010c7acb43 Tcl_WaitForEvent + 314
	22  Tcl                                 0x000000010c77c5cd Tcl_DoOneEvent + 274
	23  _tkinter.so                         0x000000010c5b0a70 Tkapp_MainLoop + 351
	24  Python                              0x000000010bee51ab PyEval_EvalFrameEx + 27003
	25  Python                              0x000000010bede63a PyEval_EvalCodeEx + 1617
	26  Python                              0x000000010bee8e12 fast_function + 117
	27  Python                              0x000000010bee50d0 PyEval_EvalFrameEx + 26784
	28  Python                              0x000000010bede63a PyEval_EvalCodeEx + 1617
	29  Python                              0x000000010beddfe3 PyEval_EvalCode + 48
	30  Python                              0x000000010bf01c04 run_mod + 53
	31  Python                              0x000000010bf01ca7 PyRun_FileExFlags + 133
	32  Python                              0x000000010bf017f8 PyRun_SimpleFileExFlags + 702
	33  Python                              0x000000010bf12eba Py_Main + 3094
	34  libdyld.dylib                       0x00007fffa5044255 start + 1
	35  ???                                 0x0000000000000002 0x0 + 2
)
libc++abi.dylib: terminating with uncaught exception of type NSException
./bCNC: line 9: 75083 Abort trap: 6           ${PYTHON} ${DIR}/bCNC.py $*
```

It seems to happen randomly, sometimes when connecting to the CNC, sometimes when just moving around in the gui. On Sierra 10.12.1.

Any advice as to fixing/tracking this down? For an error that happens several times an hour for me, I can't find instances of anyone else having this issue...

vlachoudis, 2017-03-28:
I have the feeling is more a python on Sierra issue rather of bCNC. I am sorry but I cannot be of much help here.

vlachoudis, 2017-04-11:
I am closing the issue, since it doesn't seem to be bCNC related, however if you find a solution please let us know

closed 2017-04-11 by vlachoudis