vlachoudis/bCNC #550: bCNC spindle override ERROR?

viewsat, 2017-03-26:
When sending gcode, such as changing the spindle% PWM output will have an incorrect action.
Use the oscilloscope to see PWM output.
When <100 will automatically continue to change to a minimum, when> 100 will automatically continue to change to the maximum.

See my youtube video :
https://youtu.be/5Dt0TNT3uTE

BlueOrangeLive, 2017-03-26:
@ viewsat : Maybe look in bCBC Settings Tools / Config and Tools / Controller spindle speed adjust.
Greetings Jürgen

Heszu, 2017-03-27:
@BlueOrangeLive : ??? and ???, How stop the drifting ???

vlachoudis, 2017-03-28:
@viewsat I think I found the problem. Could you check now

viewsat, 2017-03-30:
@vlachoudis I tested the spindle override function is no problem, thank you very much!

vlachoudis, 2017-03-30:
Thanks for confirming

closed 2017-03-30 by vlachoudis