vlachoudis/bCNC #551: Autolevel issue with G91

ghoyler, 2017-03-29:
Autolevelling has a problem with G91 mode (relative coordinates).
It splits up movements assuming G90 mode (absolute coordinates) resulting in too long travels.

For example, consider the following G-Code (straight line of 10 mm):
G91 
G1 F200 X10

bCNC splits this up into the following segments:
G91
G1X0Y0Z0.031F200
G1X5Y0Z0.025
G1X10Y0Z0.019

Clearly this is wrong as it leads to a 5+10=15 mm movement in the x direction.
It should be 5+5=10 mm.

PS: I am using bCNC version 0.9.8

vlachoudis, 2017-03-29:
Thanks, indeed, this is an error. The expansion is performed in absolute coordinates, converted from the relative ones and then are written as such. In my tests it was working since the G91 was on a same line with G1..., so it was never written when the line was expanded from the autolevel. But if it appears in a separate line, it is passed to the controller. I will remove the G91 from the stream if autolevel is applied, and generate a code only in absolute coordinates.

ghoyler, 2017-03-29:
Perfect, thanks!
--Gernot

vlachoudis, 2017-03-29:
I'll Leave it open until it is fixed

Harvie, 2018-11-06:
Is this still an issue?