vlachoudis/bCNC #552: Proposed of No "Z" axis movement when in LASER mode

Giochia, 2017-03-29:
In case of a twin head CNC machine, it would be useful to have no Z movement when the bCNC is in LASER mode. This would save time in having to edit long lines of code when passing from mechanical Spindle to Laser use, with the identical Gcode item.
Alternatively, if movement is not easily removable, it could be sufficient to turn off Laser during "Z" axis movements, so that the Laser beam does not insist in the same spot, creating a hole, while Z is moving.

Regards, Giorgio

Harvie, 2017-05-15:
isn't Z axis used for focusing laser when cutting thick material?

Giochia, 2017-05-15:
I do not think so, laser gets focused by focusing lens. In fact, pure laser machine do not have a Z axis movement at all. 
In any case, Z axis would be moved once, if it was for adjusting focus, but not during the job.

Harvie, 2017-05-18:
there are lots of lasers that don't have motorized focusing lens. so focusin using Z axis is obvious solution as most machines have Z axis. plus it's waaay easier to calibrate. might not really be an issue for very powerfull co2 lasers, but for 1-20W solid state lasers you might need to do cut in several passes with different height, pretty much as you do milling.

Giochia, 2017-05-19:
Are you sure, Mr. Tomas, that you have done practical test on this subject?  Maybe you have some knowledge that goes beyond usual users, if so I would be interested to know more.

Focusing of laser, as already mentioned, usual only occurs at beginning of job and not during the job itself. If, however, a focusing during the job was required (is it very unlikely), this should be done in absence of laser activity. While as it stands now, it would be impossible to do, as the laser stays on, and it would work (burn) too much in one point, while the Z axis is moving.

So, if your remark was correct, then we require the laser to be off during any Z movement. But instead, since Z movements are not useful during work, they seem to be a waste of time, specially if the Z movement is done in a slower stepping speed.

Thus, the best solution, would be to kill the Z movement when in laser mode or, to satisfy your requirement to have the possibility of Z movement for focusing purposes, it could be obtained by having the "kill Z movement" flag on the setup selection. So anyone will have the possibility to have or not to have this feature.

Harvie, 2017-05-19:
@Giochia if your stock material is eg. 2cm thick you will never have the laser focused trough whole thickness of this material. It might be problem especialy for cut surface and for lasers with lower power. That's why people might want to cut in multiple pases. And if you are eg. cutting 2cm board in 2 passes it makes sense to lower laser (and it's focal point) 1cm between passes.

Giochia, 2017-05-19:
I'm sorry, maybe I did not explain properly. By adding this feature you will not loose anything, you have added a feature and kept the ability to do what you are describing. If the feature is flaggable you will have the choice to do what you want, just do not flag it when you go in laser mode and you will keep your Z movement, and try to focus with it, where is the problem !?

Giochia, 2017-08-15:
The current laser mode turns off the laser when the gantry performs rapid movements to position to start of work position point. The same would be good if it applied to Z movement. If Z cannot be ignored, than the alternative solution would be to kill laser beam while positioning the Z axis just in the same way. Otherwise, while Z is moving, the laser makes a big hole in the work-piece.

Harvie, 2017-08-15:
@Giochia That would absolutely make sense. Even in my use case. If you use Z to refocus the laser to cut lower layer, you probably don't want it to be lasing during refocusing. This is something that i'd like to see in bCNC.

Giochia, 2017-09-08:
Today I needed to use the laser CNC for writing a logo on wood, and it turned out a bad job, because the gcode was originally written for a cutting tool, so all the time the Z axis was positioning, the laser cut great beg holes in the wood. I need to manual edit the gcode and remove all of the z movement, bu they are hundreds. I may just write the logo by hand, it takes the same time.

lalo-uy, 2017-09-08:
May be you can write a script to filter the G-code

Harvie, 2017-09-09:
Laser should be turned off while doing Z movement in laser mode. If not it's surely bug.
If you want to prevent Z movement, then you can probably set safety height to 0.

Giochia, 2017-09-10:
Hi, the laser does not turn off while doing Z movement, this is the reason for opening a ticket in the first place !
Cannot set height to zero for I have a twin head CNC machine,and often the same job is to be used for both, laser and carving.

Harvie, 2017-09-13:
@Giochia you have to reset zero in controller before starting job! nothing should crash into workpiece given that you laser focal point is lower than your endmill / milling head.

Giochia, 2017-09-13:
Sorry, I believe I cannot make myself understood. There is NO CRASH at all. It is nothing to do with crash. If you did try to use a laser CNC you may understand what I am trying to say. It is really extremely easy. Just need a menu line to turn OFF laser DURING "Z" axis movement. It is so simple that I cannot, for the life of me, accept that nobody understand it.

Harvie, 2017-09-13:
@Giochia i understand, but if you set safety height to 0, there will be no Z movement. However i don't know how to set this in bcnc. Maybe it's just not supported :-(

Harvie, 2019-07-14:
I think the solution is setting zero here:

[image]

That way your CNC will always be in Z0 (unless your g-code specifies Z movement).