vlachoudis/bCNC #553: Homing Problem

Vnaf, 2017-03-30:
Hi

I use bCNC in a Raspberry pi 3, with Arduino UNO running GRBL 0.9 for drive my CNC router.

I just installed limit switches in my machine, 2 per axis, in NO configuration, with 4k7 resistors from Limit pins to +5V.

The switches work, as I push one of them I receive a Hard limit alarm in GRBL. But when trying to do a homing cycle, the machine doesn't stop when the Z limit switch is hitted.

I've been tweaking the configuration of GRBL, I can control the homing direction of each axis, I can invert the pins, but I don't find any option about homing other than Enable homing, which of course is activated.

I'm stuck on this, as I'm almost sure it's a software configuration problem, but I just can't figure out what's happening.

Any idea about what could be causing this?

Thanks

lalo-uy, 2017-03-30:
Are you aware of the Z limit pin change to accommodate pwm spindle ?
If I remember correct pins 11 & 12 were swaped.

lalo-uy, 2017-03-30:
Also put a 100nF cap from the limit pins to ground to prevent noise false
trigers.

2017-03-30 10:43 GMT-03:00 Eduardo Spremolla <e.spremolla@gmail.com>:

Vnaf, 2017-03-30:
Hi

I used pin 11 for Z axis switch. But It works as Hard limit switch so I would think that's not the problem, anyway I will try now.

I'll also try with 100nF caps to gnd.

Thanks both for your answers!

Vnaf, 2017-03-30:
Update: Actually I'm using pin 12 for Z axis.

No changes with 100nF caps to gnd :(

Heszu, 2017-04-01:
"I used pin 11 for Z axis switch. But It works as Hard limit switch "

Did You press manually, activating the switch, or by jogging the spindle ??
Does it actully make contact when reaching end position??
Directions are right ?

A short scheme how You connected limit switches ..

WBR
H

Vnaf, 2017-04-04:
Hi

Sorry for the delay in the answer, I've been a few days out of the shop.

"Did You press manually, activating the switch, or by jogging the spindle ??"

I tried both, pressing the switch by hand and letting the spindle push it. Neither way worked. Right now I'm pushing the switch by hand, trying to avoid more crashes in the Z axis.

"Does it actully make contact when reaching end position??"

Yes it does.

"Directions are right ?"

No, but I think I know how to adjust it in the GRBL Settings ($23, Right?), actually I've seen it changing when adjusting Grbl settings.

A short scheme how You connected limit switches ..

I used this scheme, except for the resistor values, I used 4k7 Resistors. And also the Z axis pin, I'm using PIN12

[image]

Here you can find my GRBL configuration:

[image]

My limit switches worked this way perfectly, when I pushed one of them with the hand or the spindle I had an bCNC Hard Limit Alarm, so as Limit switches they worked.

Right now, I just have the Z axis switches connected to the Arduino, via PIN12.

If I close the switch manually (or pushing it with the spindle), bCNC gives me an Hard Limit Alarm.

I tried to connect the Z axis switches to the 3 Arduino pins simultaneously with the same results. When I press Home, Z Axis goes down, I push the switch and nothing happens, the Z axis continues going down, no matter if I keep it pushed or push it repeatedly, so I have to abort the homing cycle in order to avoid a crash.

The fact I don't fully understand is why if they work as limit switches, doesn't work in the homing cycle.

chamnit, 2017-04-04:
This is a common problem. You likely have your limit switches inverted when they shouldn't. Grbl doesn't detect a high/low state for hard limits. It uses the AVR's pin change interrupt which only detects a change. It isn't ideal, but it's the only way to do it on the AVR and keep it fool proof. This often leads to the user thinking that they have their limit pins wired correctly. 

If you use Grbl v1.1, which is recommended, you can connect to Grbl via a serial terminal and issue a '?' status report (or have bCNC show you the reports, not sure how to do this). Grbl should reply with a 'Pn:' field, showing you which pins are considered "triggered". If none are "triggered", then the 'Pn:' field won't show up at all. This is what you want for a properly configured homing cycle.

Vnaf, 2017-04-05:
Hi, thanks for the info

I'm using GRBL v0.9j.

If I understand correctly, I have to set $5 to 0, right? When I do it, just changes homing direction, but same no response on switch pressing. 

So maybe i misunderstood your sentence about Limit switches invert, maybe I have to change another GRBL Setting instead $5?

Regards

chamnit, 2017-04-05:
@Vnaf : Use Grbl v1.1. Use the status reports to ensure the limits are being read as you expect. Also make sure your coordinate system is oriented correctly.

Vnaf, 2017-04-05:
Hi, I just updated to v1.1. and realized I was using incorrect pin in Limit switches connector (not in Arduino itself, in the enclosure connector) for the Z Axis, so after going through the shame I connected it to the right pin and the magic happened. 

Then I connected the other two axis, and completed a few homing cycles at different speeds to check repeatability.

Thanks all of you for your help.

closed 2017-04-05 by Vnaf