vlachoudis/bCNC #554: Wrong values in the self-leveling routine

Thorgrlm, 2017-03-30:
Hi, I'm from Spain and I don't speak English well, I'm sorry fot that .
I have a problem with the measurements of the probes, the value is not correct, instead of being close to 0 are far above, in the images you can see. if you need more informatión say to me

this is the error
[img]http://i.imgur.com/Eo0TXnJ.png[/img]

Image
[img]http://i.imgur.com/XRIuxnF.png[/img]
[img]http://i.imgur.com/0i208Lh.png[/img]

Thanks you

lalo-uy, 2017-03-31:
Hola. I belive the error are in the units. may be the gcode is in mm, but
bCNC is assuming inches

10.15 mm =>  10.15 " = 258 mm

vlachoudis, 2017-03-31:
Can you verify that you have set bCNC to think in mm as well grbl when it reports

Thorgrlm, 2017-03-31:
Thanks you for you help.
I think I have it all in mm, I upload images of the configuration

[img]http://i.imgur.com/3VR7s5b.png[/img]
[img]http://i.imgur.com/LOZpLHn.png[/img]

Thorgrlm, 2017-04-01:
Hi, good morning, I fix the problem, I changed the firmware GRBL, I had installed the GRBL v1.1 and I change it to GRBL v 0.9j, and now it work fine

vlachoudis, 2017-04-01:
Good to now

closed 2017-04-01 by vlachoudis

Thorgrlm, 2017-04-02:
Thanks for the great work with the software, this is the result
[img]http://i.imgur.com/PV8aYNK.jpg?1[/img]

Is a test pcb to mount a pic18F97J60
I'm very happy with the result, thank you and a big greeting

Effer, 2017-04-03:
It looks very well done. Would you like to share your work parameters? Feed, depth, bits etc?

electrokean, 2017-04-03:
Yes, that looks like very precisely milled.  Nice work.
I'd suggest increasing your isolation width, else you may get shorts to that thin segment that surrounds all the pads.

Thorgrlm, 2017-04-03:
This is the first test, I think the speed of the Z axis is very fast, the tool nail into the PCB and leave mark, I will slow down the probe and test it.
I'll share everything I've done.

1º- I have modified a CNC3040 milling machine, I have changed the electronics to Arduino Uno + Pololu DRV8825 + GRBL v0.9j + bCNC
[img]http://i.imgur.com/pYiJ1ii.jpg?1[/img]
[img]http://i.imgur.com/Xe8EuXU.jpg?1[/img]
[img]http://i.imgur.com/lzAZ7Ta.jpg?1[/img]
[img]http://i.imgur.com/YBnC1I5.png[/img]

2º-The tool is a 10º V milling cutter with a tip of 0.1mm, the depth is 0.040mm, the feed rate 110
[img]http://i.imgur.com/cC3rTIr.png[/img]
[img]http://i.imgur.com/33CqpB8.png[/img]

I buyed the tool in Alixpress.
https://es.aliexpress.com/item/10Pcs-3-175mm-Tungsten-Steel-V-Shape-Carbide-PCB-Engraving-Bits-CNC-Router-Tool/32610752639.html?spm=2114.13010608.0.0.3f8V9p