vlachoudis/bCNC #556: Am I losing steps on the Z axis?

Thorgrlm, 2017-04-06:
Hi, I have a doubt, I checked that doing several times the autolevel in the same pcp gives me different values at the same test points.
Is that normal or my machine lost steps?, it's few but They are microns of difference.

Thanks fou your help

jeffmd, 2017-04-07:
Maybe or could be dirty pcb or pcb is not securely mounted.

Thorgrlm, 2017-04-07:
I will format the pc, the pcb is well secured
I already comment anything

Thorgrlm, 2017-04-12:
Hi, I have formatted the PC, and I still have the same problems, I don't know if it will be normal, I upload a photos
[img]http://i.imgur.com/anxcxwG.png[/img]
[img]http://i.imgur.com/JuGRRix.png[/img]

vlachoudis, 2017-04-14:
Could it be that you are using a high feed rate during the probing. Go very slow to check.

Thorgrlm, 2017-04-15:
I have the speed by default.
[img]http://i.imgur.com/IOJIPce.png[/img]

Harvie, 2018-10-29:
Sounds like mechanical or deceleration problem to me. Please reopen if there's problem in bCNC.

closed 2018-10-29 by Harvie