vlachoudis/bCNC #557: GRBL interface: no status, no machine position

juepro, 2017-04-06:
Hello,

I am running Arduino Uno with GRBL V1.1 and raspberry pi B with bCNC 0.9.4.

I can connect to GRBL, I can move the CNC, I can execute G-code, so all main functions are working.

BUT:
The status does not update: I shows "not connected" all the time, even after connecting successfully
If I force an alram on the machine the bCNC status changes to "alarm",After reset, the status of bCNC stays at alarm but the machine is running again. Seems that some GRBL status messages are not correctly parsed.
I can monitor the status changes on the terminal from GRBL, were all seems to be correct but bCNC does not understand.

other oberservation: The machine position is not updated and shows always 0.
I played around with GRBL configuration ($10 - Status report, mask) but was not successful.
The terminal shows the machine position, but bCNC does not understand it.

Any ideas what the problem could be?

regards,
juepro

jeffmd, 2017-04-07:
Use the latest bCNC.  bCNC didn't support grbl 1.1 until version 0.9.7.  I am running a similar setup as you with no problems. grbl setting $10 should be 1.

juepro, 2017-04-08:
PERFECT!
Thank you. All problems solved.
I was not aware of newer bCNC version

juepro, 2017-04-08:
NOW milling is fun :-)

closed 2017-04-09 by vlachoudis

MARIOBASZ, 2019-12-11:
I have downloaded bCNC on an old machine, it works with Windows XP.
Arduino connects, but bCNC interface shows not connected, does not show work or machine coordinates.
Some help? Thank you