vlachoudis/bCNC #558: Stop does not unlock in Smoothie mode

wolfmanjm, 2017-04-08:
Using the Smoothie mode talking to current smoothie edge.

Click the stop button when running a job, and it enters Alarm mode as expected, however clicking unlock button it remains in Alarm mode and I have to connect via a second console to smoothie and manually issue $X to clear the alarm mode.

olablt, 2018-06-28:
any way to fix this annoying bug :) ?

wolfmanjm, 2018-06-28:
I think the issue may be related to it sending a feed hold first, which on smoothie is not required. I have not found an easy way to fix it. with the latest version of smoothie it does seem to unlock most of the time.
This may also be an issue of the ok count getting out of sync. with smoothie it is safer to reset the ok count after a halt.

wolfmanjm, 2018-07-01:
On further examination I think the issue is that GRBL does not enter the ALARM state when stop is pressed, so bCNC is not expecting that, whereas Smoothie enters an ALARM mode on stop (control X), and needs to be cleared ($X) before doing anything else. So the fix is quite complex to make bCNC understand that Smoothie is in HALT state after Stop, unlike grbl.
Also it should not send ! (feed hold) to Smoothie on Stop unlike on GRBL where it does.

wolfmanjm, 2018-07-01:
Unfortunately Smoothie does not do control X anything like GRBL does (it does not reset for instance), and I have also been unable to duplicate GRBLs ability to do immediate feed hold (which is awesome!). Hence the reason why Smoothie enters ALARM mode on control X while it is moving. In smoothie control X is more of an E Kill switch.