vlachoudis/bCNC #560: When I start sending gcode for some time it will pause sending gcode?

viewsat, 2017-04-13:
When I start sending gcode for some time it will pause sending gcode, but wait for some time to continue sending.

Effer, 2017-04-13:
Hi @viewsat can you post your gCode? You can zip and drop it here.
Where are you running bCNC (Computer specifications, RPI version?)

viewsat, 2017-04-14:
I use PC debian linux, raspberry pi, orange pi, the results are the same.

Test  file LON.CNC,
https://drive.google.com/file/d/0B0BGzc0LgeZRYzZweFVLOUYwa1k/view?usp=sharing

Effer, 2017-04-14:
Just made a couple of test. It seems ok for me. Tested with latest commit and an old version of bCNC.
I'm still using V0.9i in my real machine. Don't know if newest version have some issue.
The code is quite heavy, lot's of small line and high Feed Rate (F3000). My old machine computer get very unresponsive but the buffer seems quite well fitted (80%).

Don't know what's happen to your side, can you try to modify the feed and slow down to 300? 
Does the stop happens always at the same point?
Can you post your grbl version and machine settings ($$)?

viewsat, 2017-04-15:
<1>I copied LON.cnc 10 times, you try again.
https://drive.google.com/file/d/0B0BGzc0LgeZRcnpFa2NRVVlGeWs/view?usp=sharing
<2>I use another three gcode sender program tests that are working normally without pausing.
<3>grbl v1.1f , machine settings
[image]
<4> orange PI PC + armbiam run bCNC cpu usage 32% .
[image]

Effer, 2017-04-15:
@viewsat , damn that Orange is faster then my pc :)
I'm busy with some jobs next days, maybe I'll make a test next week.
Just for test, can you try to run the job without the canvas rendering?

vlachoudis, 2017-04-16:
@viewsat I've just let it run on my machine, and it went all they way to the end with no stop.

viewsat, 2017-04-18:
@vlachoudis Because bCNC it will automatically pause after continuing to send gcode, i did not go to pause.

Harvie, 2018-10-29:
Please test latest version and check that you don't have any tool change or pause commands in your g-code. If there's still issue, please reopen.

closed 2018-10-29 by Harvie