vlachoudis/bCNC #561: Showing Not Connected when it is Connected - Just a question

jpagliaccio, 2017-04-14:
I am trying to find the proper command syntax to send to bCNC from custom Arduino code that is mimicking grbl (but is not grbl). Grbl and bCNC are working fine for me so I know it is just something I am doing wrong.  

I am sending Grbl 1.1e ['$' for help] and connecting okay but the message bar shows "Not Connected" in red but it shows "Connection is established with Grbl" when I click it.

I am also sending position commands in brackets but nothing displays in the X, Y or X windows. 

Thanks for the help.

vlachoudis, 2017-04-14:
Are you using the latest version of bCNC?
Also do you have set the $10 of grbl reporting to the default value

jpagliaccio, 2017-04-14:
Thanks for the quick reply. I just installed the latest bCNC and same problem. I am connected but it shows Not connected. 
Also my code is sending back Machine Position and Work Position like Grbl=3. Like this 
<Idle, MPos:10.000,10.000,0.000,WPos:0.000,0.000,0.000>.  

When I connect my code sends: Grbl 1.1e ['$' for help] and \n.
Do i need something else?

alexps750, 2017-04-19:
I am having the very same issue, latest version of bCNC running on a raspberry pi 3 does not show connected or show the coordinates. I have the $10 in the controller config in the tools tab set to 50. I don't know if this is the default value or not as I haven't touched this setting because I don't know what it does :-). If that is not the default value, what is?

alexps750, 2017-05-07:
i have solved my issue and it was the $10 value was set to the wrong value.  I set it to the default value and now all is good! :-)

closed 2017-05-09 by vlachoudis

mano1979, 2018-04-05:
> I am trying to find the proper command syntax to send to bCNC from custom Arduino code that is mimicking grbl (but is not grbl). Grbl and bCNC are working fine for me so I know it is just something I am doing wrong.<

What this user means is that he wrote his own firmware (not grbl) that he wants to use with bCNC.

I want to do the same. i have written a gcode parsing firmware for a custom project. But i want bCNC to control it. Though when i make a serial connection and send "Grbl 1.1e ['$' for help]" from he arduino to bCNC, it does not connect properly. 

So the quesion is: What does grbl send to bCNC when making a connection?

Seb66, 2018-05-10:
Hi i have the same issues,

[image]

Can anyone help?

excury, 2019-06-11:
> 
> 
> Hi i have the same issue.
> 
> [image]
> 
> Can anyone help?
> 
> I'm using a Rasberry Pi.
> Ive tried turning on and off the serial port.

I have tried the $10=1 that was propose earlier
I have tried to do a software reset  of the controller and the hard reset but it does not seem to complete.
V 0.9.14 8th Jan.
 
I am using a usb to connect to Eleksmaker Mana Briver Board.
> I'm using TTYusb0
> 
the motors don't move when command sent, the CNC machine does not work.  
If anyone could help me to solve this issue that would be awesome!

Harvie, 2019-06-19:
@excury please try latest pip (or git) version

oliverlambson, 2021-06-29:
Had the same issue. Turned out we had accidentally hit the emergency stop and it just needed to be reset. Don't forget to check the board actually has power!