vlachoudis/bCNC #563: An idea: Backlash Compensation

Fosforitofercho, 2017-04-17:
I dont know how difficult can be the programing, but is posible  like the autoleel, compensate the g code for counter the backlash .

vlachoudis, 2017-04-17:
That could be an interesting idea.

Effer, 2017-04-17:
That will interest also me.
Something to apply to all code (modifying it) or to apply at run time?

MARIOBASZ, 2017-06-07:
Hola, primero felicitaciones y gracias por lo magnífico que es y las mejoras tanto en bcnc como grbl. Sería bueno poder agregar una solución al backlash. Es muy elemental mi conocimiento de programación, y no pude saber dónde es que el programa envía la orden al Arduino para modificarlo. 
Si se puede agregar una corrección de holgura ante cada cambio de dirección del eje, y la actualización de la posición en pantalla es el movimiento enviado restada la corrección enviada, sería una solución. Estoy terminando de implementar un económico ATC (automático tool changer) ( las rutinas de movimiento sí las pude organizar) para mi spindle con tuerca ER11, partiendo de la brillante idea de Frank Herrmann con su XATC y me resultaría útil para mi CNC que tiene un backlash de aproximadamente 0,3mm. Saludos.

MARIOBASZ, 2017-06-07:
O en realidad siempre, en cada orden, envía la posicion del eje más la compensación de holgura, pero el valor de esta es cero excepto cuando hay cambio de dirección.

lalo-uy, 2017-06-07:
Mario, la mayoría de la comunidad aquí se comunica en Ingles.
El tema del backslash ha sido tratado muchas veces, pero no mucha gente se
intereso, por usar cnc con correas que no sufren de ese problema.

vlachoudis, 2018-01-29:
@X3msnake  Thanks for the links

X3msnake, 2018-01-29:
@vlachoudis 
If you manage to implemented do post back i'm also interested in this. Best regards and thank you for your great work

Thorgrlm, 2018-02-03:
Hi my friends, it would be interesting to implement a backlash corrector, for precision work it would be interesting, if you look at commercial software as the mach3 has a backlash corrector.
I make PCB and precision is very important

BananaLuke, 2019-03-10:
Hello together ;) 
Also antoher upvote for Backlash correction... This would be a really awesome feature for people working with small/cheap mills like the Proxxon MF 70.

MARIOBASZ, 2019-03-10:
It's one of the things I have in mind to implement: Backladh compensation and axis orthogonality correction

BananaLuke, 2019-03-10:
I had a look at the softwares which CNC controllers/programs are possible to have backlash compensation. There is only Mach3/4 and LinuxCNC. But this means adding parallel port to the computer and: 
1.) For LinuxCNC: additional Boot partition - no easy CAM processor. 
2.) For Mach3/4: Spending a lot of money on software.

Harvie, 2019-03-10:
I think this stuff should primarily be handled directly in motion controller.
I am not completely against having bCNC plugin for this, but for me it would be just workaround for cases where this cannot be done in motion controller.

Just for the record. GRBL already works on this, there's even patch:

https://github.com/grbl/grbl/issues/389

MARIOBASZ, 2019-03-10:
I have not seen it, the best solution is directly in grbl. I'm going to look at the patch

mkaluza, 2019-03-11:
It's there: https://github.com/mkaluza/grbl-Mega/commits/edge
I agree that it should be directly in grbl and so I did that. However it's unlikely that it will be merged upstream.

BananaLuke, 2019-03-20:
nice to see that there is something! Just for the Arduino Uno its missing. I have a spare RAMPS and 2560 here and will try it the next days...

MARIOBASZ, 2019-03-21:
Comment that this has worked for you.
I think the best option is to fix it
The second best options that bcnc send the modified code
The third option is through a plugin.
This is the option I can write, it is the least good of the three. Maybe the second option.

BananaLuke, 2019-03-21:
@mkaluza 
Thanks for the software. Just compiled it to the Arduino Mega. It seems to work. Just a short note on the Documentation: I first used the hex file that was listed further down on the github page, this was from gnea. Thus it will not have backlash compensation. I now included the GRBL Mega library from your fork in the Arduino IDE, deleted the old GRBL software (for the UNO) and uploaded it via Arduino IDE. 
Just a small side node you might want to mention on your github page so people dont get confused when using that fork  ;) 

reagards
Lukas

mkaluza, 2019-03-21:
@BananaLuke thanks for the info - I never expected anyone to be using it - that's why no hex file. You're the first (that I know of)  :medal_military: :) I was just about to ask if you need one :P 

I also have Uno code, but can't test it because my machine is on ramps. Anyway, happy milling. At some point if enough people are using it, maybe it can get merged upstream.

BananaLuke, 2019-03-27:
@mkaluza Hello again. I had to fix some hardware issues (well i thought so...) BUT ramps and Arduino are fine... Tested it with grbl cyclone Works. I also tried to upload grbl_mega from the "main" branch (gnea)...
This one doesnt work neither. 
Can you tell me how you configured the pinouts for RAMPS 1.4? Is the gnea branch and yours compatible with an Ramps 1.4? Might passing me a hex file for the uno?

BananaLuke, 2019-03-27:
A gif says more than thousand words! 

[image]

cowabunga!

10 kudos to @mkaluza so far. Lets See how it will perform during milling operation!

mkaluza, 2019-03-27:
Yo! Nice ;) let me know how it behaves during milling :) Are dimensions correct?

As for not working - gnea/GRBL-Mega/edge is fully compatible - with proper defines in config.h for RAMPS it should just work. As for Uno - I don't have one. Right now I'm just porting my code and will test it at some point when I have something set up (this might take a while since I have a lot of stuff to do and my machine is running on Mega/RAMPS right now)

mkaluza, 2019-03-27:
@BananaLuke I've just uploaded my test code (with hex file this time) with all my experimental features that I am currently using. It has quite a high step rate (~40kHz+) so it allows to use high microstepping values for smoother/more silent stepper operation. I'm currently using 1/16, which gives me around 1100-1200mm/s max speed - it's not much, but my machine wouldn't go faster anyway. Also the latest feature - acceleration smoothing - should in theory allow higher acceleration values without loosing steps (but I can't really test it on my machine, because it still needs a lot of mechanical work)

Harvie, 2019-03-27:
> Also the latest feature - acceleration smoothing - should in theory allow higher acceleration values without loosing steps 

Are you using s-curve acceleration? https://www.youtube.com/watch?v=qYJpl7SNoww

mkaluza, 2019-03-28:
Hi @Harvie 
No, right now the ramp is still trapezoid, but before it had piecewise constant speed(due to the way stepper control alg works in grbl) - now it interpolates speed between segments. It's not linear, but still better than nothing (and it is computationally cheap).

But thanks for the heads up - I had an interesting read last night about that stuff :) Not sure how to put in grbl yet though because the algorithm seems to be a bit complex and I don't yet know why. When I understand why it works the way it does I'll be able to modify it more easily.

If you wish, you can jump in :) There's also cutter radius compensation on the table :) and 5 axis :)

Harvie, 2019-03-28:
> If you wish, you can jump in :) There's also cutter radius compensation on the table :) and 5 axis :)

Well. I am only interested if we manage to get this upstream to original atmega328p GRBL.
Because i don't have plans upgrading to arduino mega. If i ever upgrade my motion controller, it will surely be one of these cools 32b MCUs that are everywhere nowadays (eg. STM32 or ESP32)...

mkaluza, 2019-03-28:
Doubtful... grbl maintainer is rather unwilling to do so. Besides, Uno is very low on flash/ram already, so unless there are some code size reductions, much more won't fit there anyway.

There was a mention about porting grbl to Arduino Due, but that's it for now. And I've seen somewhere a port to some STM32. Maybe at some point I'll look into it myself.

Do you have any particular controllers in mind? Anything with open source firmware in particular?

Harvie, 2019-04-02:
> Doubtful... grbl maintainer is rather unwilling to do so.

Even when it's disabled by default?

> And I've seen somewhere a port to some STM32.
> Do you have any particular controllers in mind? 

GRBL author is working on HAL, which will enable GRBL to be easily ported to other platforms. They are doing this, because they want to support 32b platforms. So i guess if we wait half a year or so, there might be GRBL for STM32...

mkaluza, 2019-04-03:
> > Doubtful... grbl maintainer is rather unwilling to do so.
> 
> Even when it's disabled by default?

The argument was that if someone has backlash then he should fix his machine, and that if he wants it fixed in software, then let either the control software do it or some gcode processor... (he wrote somewhere that he's using gt2 belts, so he actually might not be concerned with backlash, but all screw drives have backlash and you can't really help it). IMHO the argument is not valid as backlash is 'a machine thing' and as such it should be handled by machine controller, not some software, but hey - it's his project and it's all up to him.

> GRBL author is working on HAL, which will enable GRBL to be easily ported to other platforms. They are doing this, because they want to support 32b platforms. So i guess if we wait half a year or [...]

Actually there already is a port of GRBL to STM32: https://github.com/dungjk/grbl-stm32
I'm waiting for my STM32 stick to arrive and I'll start testing it. Half a year is a lot of time... looking at GRBL adding a HAL layer shouldn't be much of a problem... few evenings, yes, but not more... But I haven't seen that in any of his repos, so I can't really tell much about it.

Harvie, 2019-04-03:
>  But I haven't seen that in any of his repos, so I can't really tell much about it.

Code was not published yet, because there's some cleanup left to do. however there's discussion: https://github.com/gnea/grbl/issues/374

mkaluza, 2019-04-03:
thanks @Harvie . but you know - commits, or it didn't happen :)

Harvie, 2019-09-07:
BTW have you seen new drag knife plugin that i've included in bCNC? Sometimes i wonder if that would be good starting point for backlash compensation. Maybe the machine with backlash works similar to drag knife... At least the concept of actual cutting point laging bit behind the toolpath might be the same.

But there are few things that will be different in backlash compensation:

0.) You will want to apply backlash compensation on 3D g-code, while dragknife plugin currently only works with 2D toolpaths.
1.) lag will be different in X and Y (you can at least compensate for the smaller from the two by seting it same for X and Y)
2.) there will probably also be backlash in Z? (well... maybe the gravity corrects the backlash)
3.) there will be no pivot points (easy to disable in plugin)

MARIOBASZ, 2019-09-07:
@Harvie, I have not looked at the code of your complement drag knife
I understand that it must be simple to generate backlash correction with a complement, in each change of direction add a line with the compensation on that axis; orthogonality of the machine can also be corrected.
However, the ideal is
correct from grbl.
Second option: instead of complementing, sending is modified when sending the gcode

Harvie, 2021-01-07:
To get backlash compensation please consider using grblHAL instead of GRBL. It will probably not run on atmega, but upgrade is cheap (less than $10). grblHAL can be compiled with backlash compensation support (as well as other improvements to original GRBL). It is now officialy supported by bCNC.

https://github.com/terjeio/grblHAL/

closed 2021-01-07 by Harvie