vlachoudis/bCNC #566: Milling head moved first to the false direction

danielpcb3d, 2017-04-20:
I'm a bcnc newbie and i'm very happy with it but i have a little problem, which i didn't get fixed. I've built a pcb milling maschine based on a grbl flashed board which i control with bcnc. For every milling prozess i use the autolevel function. After that i press "Start" and the milling head moved first to the false direction, comes back to his previous position and start milling. What i'm doing wrong?

vlachoudis, 2017-04-21:
there should be something in your gcode like a g28 to go to a reference position first

danielpcb3d, 2017-04-21:
Thanks a lot for your reply. How can delete this gcode or can i configure in bcnc that the milling maschine have to start at the given position?

danielpcb3d, 2017-04-30:
Has nobody an idea how to delete the reference position?

vlachoudis, 2017-05-09:
From what I know, you cannot delete, but at least you can define to something meaningful.
Or you have to modify your g-code generator not to use this command

closed 2017-05-09 by vlachoudis