vlachoudis/bCNC #567: Pause and Start problem

quangvi, 2017-04-26:
Hi admin @vlachoudis   , i 'm using GRBL , if I press connect to Grbl, bCNC display Connet 
if i press Start Program and press Pause late, button Start not execute again
if I press Reset on bCNC first ,bCNC display Idle ,  button Start and Pause is normal , something wrong ?

user012flo, 2017-04-26:
Try to press Pause again, to continue your program.

quangvi, 2017-04-26:
no no, when the bCNC display " Connected" , @user012flo   I Try to press Pause again, but its nothing execute , i resolve it's now , in source bCNC , when bCNC connect "COMxx", it's send " Reset" the bCNC go to "Idle" before Start program

Harvie, 2018-02-27:
Same problem. Once i pause, i can't unpause.

neilgill, 2018-08-09:
I tried double clicking Pause, then double clicking Pause again, wait between each double click, after two or three goes, my machine resumed. It seemed to work consistently, several double clicks gets it going.

Harvie, 2018-10-29:
Seems to be fixed

closed 2018-10-29 by Harvie

luisca95, 2018-12-13:
la primera vez que le di un solo click no me funciono lo que hice despues fue solo darle doble click al boton pausa y funciono volvio a correr el gcode

Harvie, 2018-12-14:
No tengo dinero. No no no no.
Try upgrading to latest bCNC.