vlachoudis/bCNC #569: Tool  change broken?

TheArcher1980, 2017-04-29:
When using the tool change via TLO, the machine first moves the Z-axis down to the tool change location and then to the x/y coordinates (possibly bad?). After successfully probing and setting the TLO, the machine moves back to its x/y start location while in G90 mode and then the Z-axis is moved back up in G90 mode with the z-value after starting. This then crashes the axis into the limit switch, as my machine first moves into the upper left corner with Z raised to the top.

TheArcher1980, 2017-04-29:
After some more experimentation, it seems this happens if the tool is longer than the probe and the TLO is a positive value. If the TLO is negative, it works just fine.

vlachoudis, 2017-05-01:
Can you try to change the type of Tool change policy

TheArcher1980, 2017-05-04:
I found the "bug". When starting a tool change, the work coordinates are saved, it then moves in machine-coordinates to the probe location, probes, and then moves back to the work-coordinates. If the z-axis is at machine 0 and work 0, it would be a negative work value at machine 0 after probing with a tool longer than the probe.

vlachoudis, 2017-05-09:
Thanks for looking into this. I will have to check when I will be back

berkakinci, 2017-11-26:
I can confirm this bug on current master branch as of yesterday.
In my case GRBL throws Alarm 2, so no crash.

berkakinci, 2017-11-26:
After inspecting the code, I see the problem happens around:
https://github.com/vlachoudis/bCNC/blob/8636836c54428df73892d4e54233af5af9ec36a4/CNC.py#L1714
Because I started the program from the home state (0,0,0), it restored to a new (0,0,0) that was outside the machine bounds.

User Workaround:  Before starting the program, jog to a lower Z (Like -40mm).  During the program, the tool remains close to the work, so this is possible but less likely to be a problem at other tool changes.

vlachoudis, 2017-11-29:
Can you try with the latest master.

Harvie, 2018-10-29:
closing for lack of response, reopen if needed.

closed 2018-10-29 by Harvie