vlachoudis/bCNC #570: Midi2cnc error

LuisDiazUgena, 2017-05-01:
Hi!

When trying to parse a .mid song (random download) there is a error in the console:

`Traceback (most recent call last):
  File "C:\Users\Ciencia2\Downloads\bCNC-master-20170501T093403Z-001\Utils.py",
line 397, in __call__
    return self.func(*args)
  File "C:\Users\Ciencia2\Downloads\bCNC-master-20170501T093403Z-001\ToolsPage.p
y", line 1449, in execute
    self.tools.getActive().execute(self.app)
  File "C:\Users\Ciencia2\Downloads\bCNC-master-20170501T093403Z-001\plugins\mid
i2cnc.py", line 244, in execute
    duration = ( ( ( note[0] - last_time ) + 0.0 ) / ( midi.division + 0.0 ) * (
 tempo / 1000000.0 ) )
UnboundLocalError: local variable 'tempo' referenced before assignment
`
What can I do to fix this? I've been looking for a answer online and can find anything on how to fix this.

Thanks

CarlosGS, 2017-05-01:
Hi Luis, as far as I can tell it must be a problem with that particular
.mid song, it does not have a defined tempo.
Maybe you can try with other .mid files?

LuisDiazUgena, 2017-05-01:
Hi!

It was a problem with Ableton (the program that we were using) With other program there was no problem.

closed 2017-05-01 by LuisDiazUgena