vlachoudis/bCNC #571: Proposal: Command Extension

roman-romaniuk, 2017-05-02:
Hi All,

Was wondering if there was any interest in extending the current command line interface to be able to accept commands (and return results) over a socket.  Would make doing batch production runs a lot easier.  I'm willing to take on the effort.

Regards,

Roman

vlachoudis, 2017-05-09:
You can already send commands though the web interface, see the wiki for more information. There was a discussion to use websockets for making it faster, or to create a tcp/ip server in order to allow external programs to communicate. 
Any help is wellcome
V.

Harvie, 2018-10-29:
We already have remote HTTP API. Please create new issue, if you miss some feature.

closed 2018-10-29 by Harvie