vlachoudis/bCNC #572: [Feature request] Add a Coolant on off switch

onekk, 2017-05-04:
Hi to all, I'm restarted used my CNC after a while and loaded the new grbl 1.1 and the corresponding bCNC 0.9.9 (last master on github).

I want to add a coolant On (M8) and Off (M9) button, (maybe a toggle).

I have investigated the code i found the RunGroup and modified it like this
`
#===============================================================================
# Cool Group
#===============================================================================
class CoolGroup(CNCRibbon.ButtonGroup):
	def __init__(self, master, app):
		CNCRibbon.ButtonGroup.__init__(self, master, "Cool", app)

		b = Ribbon.LabelButton(self.frame, self, "<<Run>>",
				image=Utils.icons["start32"],
				text=_("Start"),
				compound=TOP,
				background=Ribbon._BACKGROUND)
		b.pack(side=LEFT, fill=BOTH)
		tkExtra.Balloon.set(b, _("Turn On Coolant M8"))
		self.addWidget(b)

		b = Ribbon.LabelButton(self.frame, self, "<<Stop>>",
				image=Utils.icons["stop32"],
				text=_("Stop"),
				compound=TOP,
				background=Ribbon._BACKGROUND)
		b.pack(side=LEFT, fill=BOTH)
		tkExtra.Balloon.set(b, _("Turn Off Coolant M9."))`

Now the problem is: how to make it execute the M8 and M9 commands (apart from finding the appropriate icons and add to bCNC)

Some hint are appreciated.

Many Thanks and Regards.

Carlo D. (onekk)

vlachoudis, 2017-05-09:
The RunGroup maybe is not the most adapted one, since it sends a global message that it is intercepted by the master event loop and then it is forwarded to the appropriate action.
Better look the code inside the DROFrame that sets to zero the WCS.
like the button self.xzero = Button(... command=self.setX0 )
and in the method you are going to create, use the self.sendGCode("M8") to send the code

onekk, 2017-05-09:
Many thanks Vasilis,

A little thing, if i found some icons, for the buttons, there are "standard" dimensions to use.

I'm thinking of adding a modified "shower" icons (coolant) or maybe a fan icon (vacuum).

vlachoudis, 2017-05-09:
I use 16x16 for the small icons and 32x32 for the big ones.

Harvie, 2018-10-29:
We already have coolant buttons. Please provide pull request if you want to upload new icons.

closed 2018-10-29 by Harvie