vlachoudis/bCNC #573: cam doesnt work with opencv

user012flo, 2017-05-04:
Hi to all, I have a problem with my camera in bcnc.
If I switch on the camera in bcnc. It shows me only a black picture.
[image]
If I execute cv2.VideoCapture(0).read() in python. It gives me following output:

[image]

In other programs like skype the camera works fine.
The same with another camera.
Can someone help me?

Effer, 2017-05-06:
Hi @user012flo ,
that's quite strange. It shouldn't allow to turn on camera if something is missing.
What OS? Can you verify if you have PIL installed?
`from PIL import Image, ImageTk`

Cheers

user012flo, 2017-05-07:
Hi @Effer,
I use Windows 7 64bit.  I have installed PIL.
In python I can import Image, ImageTk from PIL.

Greetings

Effer, 2017-05-07:
Let's try this:

``` python
import numpy as np
import cv2

cap = cv2.VideoCapture(0)

while(True):
    # Capture frame-by-frame
    ret, frame = cap.read()

    # Display the resulting frame
    cv2.imshow('frame',frame)
    if cv2.waitKey(1) & 0xFF!=255:
        break

# When everything done, release the capture
cap.release()
cv2.destroyAllWindows()
`
You should see a preview of your camera, first camera for opencv.

user012flo, 2017-05-07:
Hi @Effer thanks for your effort.

unfortunately I also see a black screen.
Frame contains a lot of zeros, like in my first post.

Effer, 2017-05-07:
That's not so bad, it means the problem is out of bCNC. 
Let's try to use another index for the webcam,  change 

`cap = cv2.VideoCapture(0)`
to
`cap = cv2.VideoCapture(1)`

user012flo, 2017-05-07:
I have already tried all webcam indexes. I also tried cv2.VideoCapture(-1), but the problem still exist.

user012flo, 2017-05-08:
I tested my camera on a virtual ubuntu 16.04.2 and there it works like a charm!

Because I want to keep windows because of estlcam, I will make a parallel installation of ubuntu.
I hope to solve the problems with Windows in the future.

Effer, 2017-05-08:
Sorry @user012flo , it seems like windows has poor support for your camera with opencv. 
Seems also you're not alone out there. There's a lot of forum with the same problem you're facing. But very little solution. 
Maybe you could try with another camera.

closed 2017-05-09 by vlachoudis

user012flo, 2017-05-11:
Meanwhile, I have tested two more cameras. The same problems appeared.
I also tested a another win7 pc. Curiously, the cameras worked there for a while.
Meanwhile, they are no longer working with opencv and win7.

I stay with ubuntu (btw lubuntu). Everything works great!
Thanks for your help!