vlachoudis/bCNC #575: Manual Feedrate Change does not seem to work

alexps750, 2017-05-07:
I am running GRBL 0.9 with the latest version of BCNC and I cannot get the manual feedrate slider bar or the value input box to adjust the Feedrate of the machine. Am I doing something wrong or not?  it kind of work in rapid mode by slowing rapid down by 25% or so but i want to do some manual machining and adjust the regular feedrate to a few inches per minute and cannot do so.

Effer, 2017-05-07:
Hi @alexps750 , you should consider upgrading to latest GRBL (1.1f) where the override mechanism has been improved. 
The old one (V0.9 and so) have some big latency (mostly depends on type if gcode and grbl buffer) and often takes some times (seconds to minutes) to apply.

Be sure you're using the correct grbl version in bCNC (Grbl-V0):

[image]

vlachoudis, 2017-05-09:
You can also enable the feed appending command from the command line "feed on"
it will append the last feed command on every line, therefore it will make the feed override a bit more interactive. However to fully experience the feed override, it is strongly recommended you upgrade to V1.1 of grbl

alexps750, 2017-05-09:
I have upgraded from to v 1.1 with no effect. Not sure why the issue got
closed already without anything getting resolved. Federate changes do not
work from terminal commands or from slider bar. Only way I can modify
federate is by writing goods in the editor window and running it.

vlachoudis, 2017-05-09:
You have grbl v1.1 and the latest master of bcnc? Can you check if the controller appears as Grbl and not Grbl-V0. Also verify that you are using the default value on the$10 reporting string of grbl.

alexps750, 2017-05-09:
I am reporting $10, just flashed the newest GRBL 1.1 to my arduino uno, and
the latest bcnc, still no feederate control.  As a side not the own spindle
enable / speed doesn't work either I've just been using a pot on the
spindle control board.
, but that is another issue.

Thank you

alexps750, 2017-05-10:
Sorry if I came off as b**chy didn't mean to.

Are other things that I should have set up between GRB and bcnc in order
for things to communicate better along the serial connection?  I am running
the software on a raspberry pi3 I BTW, I don't think i mentioned that
before.

jeffmd, 2017-05-10:
alexps750: I have a similar setup as you (raspberrypi + arduino uno) and I have had no problems with feed override using the UI or through command line.  As vlachoudis stated make sure the controller is grbl and not grbl-v0 in the connection settings.  I did have to use the drop down selection to switch to grbl when I updated to v1.1.

jeffmd, 2017-05-10:
I also use the spindle enable/speed control with no problem.  Don't know what your spindle controller accepts as a signal but that may be the problem or you are using the wrong pin on the arduino for spindle control.

zuhaibwam, 2018-03-10:
how can i get feed rate less than 1 ,f=0.18