vlachoudis/bCNC #578: User INI not working on some Windows PCs

bluetinc, 2017-05-09:
The functions 
	iniUser   = os.path.expanduser("~/.%s" % (__prg__))
	hisFile   = os.path.expanduser("~/.%s.history" % (__prg__))

do not operate correctly on some Windows PCs.  This seems to be tied to if a 'HOME" is set on the system, and if it is, then os.path.expanduser provides the 'HOME' rather than 'USERPROFILE'

vlachoudis, 2017-05-10:
Can you try setting the HOME variable?

ppsoukup, 2017-05-10:
Hi Vasilis,

The issues is actually the other way around.  If HOME is not set, everything works as is, when HOME is set, than it fails.  HOME needs to be set on some PC's for particular programs they are running (I belive Cadence tools sets) that use the HOME setting (and HOME is not set to the user).  

I created a fork of the code, and a quick fix that corrects this, but I'm not sure if I followed the proper workflow for github as this is my first go at it.  I'm also not sure with my fix, how it would work on non-windows operating systems.  There may need to be more done to it to make sure it's cross platform compliant.  -- Peter

Harvie, 2018-10-29:
is this still an issue?