vlachoudis/bCNC #580: Coordinates

Avalonnw, 2017-05-10:
Hello,
Let me first commend on the great job you are doing here!
Unfortunately I'm struggling a lot and hope you can help me.
To start with, I've got a 3040 Chinese CNC and hooked it up to LinuxCNC. It was working fine however I needed to do some manual tool changes and couldn't find a way to do manual TLOs. That contributed into my decision to switch to grbl mega and bcnc.
So far I spent a week or so with next to no result. Every time it works it looks like a fluke.
Here's what I noticed:
1. After homing cycle Mpos sets (depending on where home is) to -1,-1,-1 or -largeX,-largeY,-1. Whenever you try to set reference 0 (fusion 360 postprocessor likes to do that) it tries to move Z to 0, hits the switch, goes into alarm state. The only way around is a bad one - home first and reset arduino. It sets Mpos to 0 on all axis.
2. My favorite. Any time you try reaching EXACT soft limit, you trip it. Sorry, not every time, about once in 10 tries it actually works. So, going from z=0 in Mpos to z=-20, zeroing G54 at that point and executing G0 Z20 trips soft limit 90 percent of the time. It looks like this happens more often if the values are not nice and round but have a few decimals.
3. Tool probing (when going to probing location) moves Z down first and then moves x and y. After probing it moves X and Y first, then Z. Shouldn't it be other way around? Also, a bit of a manual for tool probing would be appreciated (I'm aware of that tiny page on wiki).

I think one of the possible solutions would be introducing proper axis setup like it's done with LinuxCNC. User sets minimum and maximum on every axis and where the home switch is. Whatever gcodes are, they get converted on the fly so grbl understands it. All tool offsets, coordinates transformations etc happen in software.

Now, a bit of rambling. You can skip it, it's just me venting frustration out.
I don't understand why grbl and grbl-related developers are so attached to old poorly developed technology. All these 'historical' reasons for negative workspace, stupid G-commands that some drunkard came up with in 60es... Come on, that is some bull's manure and everybody knows that and does nothing. LPT port, really? I understand there should be a standard, even if it's as old as my granma's toenails, but surely software-wise it could be made easy and user-friendly? I mean, do we need all these work offsets on the actual controller? Why not just let it run some simple codes in one user-defined area (like G0, G1, arcs and external equipment control) and set everything else on a PC? We need to control one single point in 3d space. It can't be that hard, can it? Or is there a fear that machinists of old would become redundant?
Sorry for that.

TheArcher1980, 2017-05-10:
I'll answer your questions as good as possible for me. 1: GRBL issue, look into the configuration files for GRBL, it's no issue of bcnc.
2: This is a GRBL issue as well. But basically, this is an issue of the stepper motors. If moving to exactly -100, you might end up at -100.001 or -99.999, as your steps per mm are most likely not an even number.
3: Already filed as a bug.
bcnc is just a front-end to the controller (GRBL), not a controller itself as LinuxCNC

chamnit, 2017-05-10:
1: If you have homing switches, you need to learn how to use work coordinate systems. The machine coordinate system is simply a consistent reference space. All you need to do is set you work coordinate frame as zero anywhere you want. There are six of them. 

2: Grbl tracks position in terms of steps. So if you move one step beyond the soft limit or equal to it, it will trigger.

chamnit, 2017-05-10:
I also suggest learning about CNC in general. There are usually good reasons for these things that an "old drunkard from the 60's" comes up with to exist. They mostly have to do with making production operations easier, quicker, more consistent, and most importantly reduce user error/crashes/personal injury. While some things with CNC are admittedly dumb, they are time tested and will help you in the long run.

Avalonnw, 2017-05-11:
TheArcher1980, thanks. I could not find this setting in grbl code so assumed it was a bcnc issue. I'll have to dig through again.
bcnc does so many things already it might become a controller soon.

chmnit, I do have limit switches and I learn extensively every day (about CNC in general too). One of the advantages of new people coming to the industry is they bring some insight into subject. What was true some time ago is not necessarily true now. Generally technology has evolved quite a lot, as well as standards, quality control, etc. Few years ago personal 3D printing was not possible, now it's on par with regular printers. CNC was something you could find in big machinists shops, now you can build one for less then a 100 bucks. People tend to cling to well-established tech and standards, but these people would eventually go, and old standards and tech would stay. It's already a problem to find a LPT-enabled motherboards, for example. So why not revamp the whole thing?
Sorry, it's probably not a place for a discussion like that.

chamnit, 2017-05-11:
@Avalonnw: Agreed. That's why Grbl supports a limited set of gcode commands, which are the basic ones necessary to operate a machines. 

Why keep old things around? Because they don't need reinventing. Or they are so common that literally hundreds of thousands of people, young and old, all use them. I liken this to programming languages. Some are old like C, but they are still used to this day becuase of the great tools and compiler built upon them. Another example is JavaScript. It's a very powerful lambda language but is still used regardless of all it's warts. You'll soon learn that while some things aren't perfect, it'll take way more work to convince thousands of people that you need to do it differently. There really has to be a compelling reason and with gcode, there isn't that much of one since CAM just translate everything for you.

vlachoudis, 2017-05-11:
Compatibility is to keep the mistakes of the past.
Reinventing something new is not impossible, however the (basic) g-code standard is compact, simple, easy and it does well the job it is intended to so.

Once you have homing configured properly you don't need to deal any more with the MPos

For fun:  https://xkcd.com/927/

vlachoudis, 2017-05-11:
@Avalonnw @TheArcher1980 I've missed the (3) Tool probe, this is not a bug, it is done on purpose. The motion of the tool probing is the following
1. First raise to the tool change height which should be the safe height for all g0 motions
```g0 z [tool change z]```
2. Then move X/Y to tool change location
```g0 x [tool change x] y[tool change y]```
3. At the same height of [tool change z] which is the safe Z height move to the
probing location in X/Y
```g0 x [tool probe x] y[tool probe y]```
4. Lower the gantry to approach the switch, to the probe height just above the switch
    Here maybe comes the confusion that I assume the switch is lower than the tool change location.
**```g0 z[tool probe z]```**
   In my setup I have the switch mounted on my machine-work-surface which is always lower than the working piece. So if I was inverting the steps 3,4 it wont work.
5. Probing the switch
6. After the probing go back with an intermediate location the tool change position
 Raise to the tool change height. As I mentioned this should be the safe height for movements
```g0 z[too change z]```
7. Move horizontally to tool change location
```g0 x [tool change x] y[tool change y]```
8. Restart spindle
9. Move horizontally to the initial X/Y location
10. Move vertically to the initial Z location

The condition that should be respected is that the [tool change z] > [tool probe z].
One possible correction would be that if I see that the [tool probe z] is higher than [tool change z]
then invert the steps 3 and 4. However what to do for the steps 6 and 7 in this case?

Avalonnw, 2017-05-12:
Can you do
all in G53
1. G0 Z0
2. G0 x[tool change x] y[tool change y]
3. G0 Z[tool change z]
4. G0 Z0
5. G0 x[tool probe x] y[tool probe y]
6. g0 z[tool probe z]
7. Probe
8. G0 Z0
9. G0 x[tool change x] y[tool change y]
8. Restart spindle
9. Move horizontally to 0,0 in work coordinates
10. Start the job
?

vlachoudis, 2017-05-12:
You are assuming that the "G53 G0 Z0" is the safe height, I am not sure that this is a correct approach.
Maybe the best would be to define an additional variable as safeZ, or instead of the
G0 Z0 to be replaced as G0 z[max(toolChangeZ, toolProbeZ)]

Avalonnw, 2017-05-12:
That would definitely work.