vlachoudis/bCNC #581: Memory leak on MacOS Sierra (10.12.4)

oMtQB4, 2017-05-10:
Hi there,

we are currently facing problem regarding memory leak. After running just one job, the used memory increases from about 100MB to 8.6GB.

Best regards

vlachoudis, 2017-05-10:
Could you check with a previous version to find out from when it started? Also do you mind sharing your gcode to check on Linux?

oMtQB4, 2017-05-10:
We tried on Linux and it doesn't seem to be a problem there.

I will try another version of Python tomorrow. Currently, I'm using the python from homebrew. Tomorrow I will try the stock version of MacOS.

vlachoudis, 2017-05-11:
Could you verify that it is really the memory used or the virtual memory allocated to the process. On linux the real memory appears with the "top" command as "RES" memory, while the VIRT is a bit more complicated, and it is not an indication of the real memory used. I don't know on Mac how it behaves.

oMtQB4, 2017-05-11:
On the Mac it is definitely the memory used. The system shows a warning after some usage that it runs out of memory. Some other programs do not respond anymore, etc.

I tested different versions (without an actual CNC machine connected, just some serial port) and if I load the file, start the process, stop it, start it, ... more and more memory is used (not as much as running a complete G-Code).

The versions of the repository I tested:
- Current Master
- 0.9.9
- 0.9.3
- 0.7.7

Currently, I'm using Python from MacOS Sierra (2.7.10) but there is no difference for all these versions.

oMtQB4, 2017-05-12:
Maybe, it is not a problem of Python but instead of some library that is used, e.g., Tkinter

I will soon try some things to see if the bug can be corrected.

oMtQB4, 2017-05-31:
I think that the memory leak is caused by tkinter, but I don't know how to fix it. A problem that occurred is the open file dialog. It never closes, it just is available as a second window of bCNC.

closed 2023-04-23 by oMtQB4