vlachoudis/bCNC #582: Interface suggestion

Avalonnw, 2017-05-12:
3 ideas.
First, the important stuff like BIG STATUS BUTTON is on 3 tabs only. Isn't is important enough to be on every tab?
Second, would it be possible to read comments section in header of g-file (if available) and show extended messages? Like, "Tool change - T02 - Flat End mill 3mm" or something like that.
Lastly, I don't really get the way the editor splits the code into blocks. I might be wrong, but to me it would be logical to follow comments with operations names, and create like a tree structure:
Header
    |
    |-Operation 1 - Tool 1
    |            |- block 1
    |            |- block 2
    |
    |-Operation 2 - Tool 2
    |            |- block 1
    |            |- block 2

vlachoudis, 2017-05-12:
1) The Status button is user configurable in bCNC, it is documented in the Wiki  how to change it
2) This is already done, but is using the inline comment of the tool change line and not the preceding ones.
3) The split into blocks is done following the rapid Z motion. It starts with the g0 motion before you plunge into the stock, until the moment it leaves the stock.

Avalonnw, 2017-05-12:
1. After a quick search I found how to add various panels to ribbon but could not see status button setting. Sorry.
2. Ok. I did not see that because my files are from fusion360 grbl postprocessor and do not have inline comments:
`%
(1001)
(Arduino_case)
(T1  D=6 CR=0 - ZMIN=-17 - flat end mill)
(T2  D=6 CR=0 - ZMIN=-19 - flat end mill)
(T3  D=3 CR=0 TAPER=118deg - ZMIN=-20.401 - drill)
(T4  D=2.5 CR=0 TAPER=118deg - ZMIN=-20.251 - drill)
G90 G94
G17
G21
G28 G91 Z0
G90

(2D Adaptive1)
M9
T1 M6
S16000 M3
G4 P10.
G54
...`
3. Understood. It's just a bit hard to navigate if something goes wrong and you need to find a certain piece of code. Is there a record on "last executed line before fail or alarm triggered" anywhere? Sorry if I'm asking too many stupid questions.

nealre, 2017-07-11:
Hey, i'm new and trying to learn BCNC. One of my problems is that it times out and the router just stops and if you try to restart, it starts all over again! not good. I can't fine any way for pickup where i left off? I realize you would need to give the m3,  s1000 commands to start it at the stop point, but there are no provisions for such? I guess? I tried the command line. "RUN Block {34}" no work!
My problem maybe solved another way if an Arduino or Raspberry PI could be used to control it. Has someone done that yet? if so, please direct me to the site.
thanks,
Rick
nealre990@gmail.com

closed 2018-10-29 by Harvie