vlachoudis/bCNC #583: G64 and M07 cause bCNC to stop running gcode?

nathanielstenzel, 2017-05-13:
"G17 G90 G64 M3 S3000 M7" seems to get objections from bCNC. Changing it to "G17 G90 M3 S3000" will make it work again. Does anyone have any idea why bCNC would object to G64 and M7? For those interested, I also tried the M03 and M07 variants. Yeah, I do not have mist cooling on my machine. I simply add things that a person might need. I am using a gshield.

G64 - Other - Best speed path
M07 - M-Code - Mist Coolant ON﻿

vlachoudis, 2017-05-14:
Is not bCNC that objects, rather grbl returns and error since it doesn't understand these two commands. bCNC following the suggestion of grbl author @chamnit it stops execution on any error, otherwise it can lead to a wrong job.
However if you want to tell bCNC to skip these commands you can do by setting the following in the error section of the .bCNC file. Look the wiki for more informatino

[Error]
g64 = 1
m7 = 1

closed 2017-05-14 by vlachoudis

nathanielstenzel, 2017-05-14:
Thanks for the explanation.

treki1, 2019-06-22:
I enter into bCNC.ini
**[Error]
G64 = 1**
But if the comand comes with parameter (eg. **G64P0.05** ) the error appear:
"Last error: P0.05"
Is it possible to ignore the following parameter?

carl1961, 2019-06-23:
I think you need a space between G64 and   G64 P0.05     **G64 P0.05**