vlachoudis/bCNC #584: Error at End of Running Auto Level Probe

woodworkerbob, 2017-05-13:
Not sure where to post these kinds of questions, I'll do it here.  If someone will kind enough to let me know where to post other questions.

Running Auto Level scan and got an error.  Below is a copy from the transcript.

G38.2Z-6F0.3
[PRB:-107.650,-205.900,-39.800:1]
ok
G0Z2.5
ok
G0X-104.775Y-61.0693
ok
Grbl 0.9g
Run ended
2017-05-12 14:55:53.678000
Current: 125 [125]  Completed: 99% [5m29s Tot: 5m29s ]
f0.3$X
error: Expected command letter
G0 G54 G17 G21 G90 G94
ok
G43.1Z0.000

I have changed my bcnc.init file to add the Control function to the Auto Level function.  Below is my file:
[bCNC]
page = Probe
probe.page = DRO Control ProbeCommon
width = 1366
height = 705
sash = 463
windowstate = zoomed
language = en
lastcheck = 1460759055

Let me know how I can help.
Bob

vlachoudis, 2017-05-14:
This might be related with the feed override addressed in #579, please download again the latest master and try again

Harvie, 2018-10-29:
closing, please let know if that didn't helped

closed 2018-10-29 by Harvie