vlachoudis/bCNC #585: Add option for tiling using g-code procedures, loops and offsets

Harvie, 2017-05-13:
I've heard that it's possible to tile g-code by wrapping it as procedure and then calling that procedure again and again with different offsets from loop. Maybe it could be nice option for tiling feature in bCNC. However the oldskool method should still remain available as procedures might not work everywhere...

vlachoudis, 2017-05-14:
Unfortunately, grbl is not having procedures and functions, so we are forced to do the tiling at higher level

closed 2017-05-14 by vlachoudis

Harvie, 2017-05-14:
I know, but grbl is not the only interpreter... Also i think it might motivate such efforts if it's implemented on bcnc side. Shouldn't be really hard to implement... Even it might not be possible to do with current hardware that grbl uses.

vlachoudis, 2017-05-14:
For the moment in bCNC  gcode evaluation there are no functions definitions. Maybe in the future